raspibolt/raspibolt #823: [FEATURE REQ] Migration guide from Umbrel to Raspibolt

heredia21, 2021-12-06:
Requesting a migration guide. Umbrel -> Raspibolt

As an Umbrel user who would like more control over my node, I would like to migrate my node and start using Raspibolt, without having to re-download the blockchain.

VajraOfIndra, 2021-12-12:
For now, you could use this FAQ entry: [https://raspibolt.org/faq.html#can-i-update-my-raspibolt-2-to-the-new-version](https://raspibolt.org/faq.html#can-i-update-my-raspibolt-2-to-the-new-version)

Although it mentions migrating from RaspiBollt v2 to v3, the process is very similar for any kind of manual migration that required copying the bitcoin and lnd data.

However, the guidelines above are a bit too short and lack useful additional information to avoid running into issues (particulalry if a user is not yet too familiar to the command line for example). So IMHO, I do think it would be worth writing a detailed bonus guide specifically targeted at Umbrel users who'd like to migrate smoothly to a RaspiBolt setup.

@Stadicus wdyt?

Stadicus, 2021-12-12:
I'm not too familiar with Umbrel, but I know that it heavily relies on Docker images. I'm not sure if "migration" is doable, because recreating all the necessary users and directories on the RaspiBolt is basically the same as setting it up from scratch.

I fear that subtle differences in the underlying operating system can become a major support burden. And it also adds Umbrel as a dependency, with us needing to keep up with Umbrel changes.

Maybe some simple instructions on how to copy the Bitcoin blockchain and LND configuration from an Umbrel drive to a RaspiBolt drive are doable, but I'm not eager to add more (ongoing) work that distracts from the main goals of the RaspiBolt.

VajraOfIndra, 2021-12-12:
yeah, by 'migration' I just meant explaining how to copy/paste the blockchain data and LND data and other useful files on a newly created node following the RaspiBolt guide, nothing more fancy than that.

VajraOfIndra, 2022-02-12:
> Requesting a migration guide. Umbrel -> Raspibolt
> 
> As an Umbrel user who would like more control over my node, I would like to migrate my node and start using Raspibolt, without having to re-download the blockchain.

Did you manage to migrate your node? or would you need some help to do it?

heredia21, 2022-02-12:
I managed to migrate my node. Thank you.

closed 2022-02-12 by heredia21

VajraOfIndra, 2022-02-12:
> I managed to migrate my node. Thank you.

Excellent. Do you have some notes we could use to draft a rough migration guide? The key things are the absolute paths of the bitcoin and LND data folders in Umbrel. And also a list of potential other program that have important data folder to migrate (not sure which ones yet).

RooSoft, 2022-02-14:
@heredia21 Definitely interested in notes, as I am closing in on the time of my own migration.

heredia21, 2022-02-14:
I apologize for the false hope. I migrated from using one node to another, I did not migrate/transfer any data from Umbrel. I purged my Umbrel node and started over.