raspibolt/raspibolt #826: Blockchain Explorer cannot be updated v.3.3.0

Ubuntu51, 2021-12-08:
### Describe the issue
A clear and concise description of where you're stuck in the guide. What are you trying to do, and how is it not working?

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4, Odroid, ... ]
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]
 - Version: [ output of `lsb_release -d` ]

### Additional context
Add any other context about the problem here.

Ubuntu51, 2021-12-08:
I try to make an update as described in the instructions 
https://raspibolt.org/btcrpcexplorer.html

git checkout v3.2.0 (old)
when I enter git checkout v3.3.0( New Version) comes the message 

**`error: Your local changes to the following files would be overwritten by checkout:
	npm-shrinkwrap.json
Please commit your changes or stash them before you switch branches.
Aborting`**

Do I enter npm install 

**npm WARN bootstrap@5.0.2 requires a peer of @popperjs/core@^2.9.2 but none is installed. You must install peer dependencies yourself.

audited 482 packages in 8.297s

43 packages are looking for funding
  run `npm fund` for details

found 15 vulnerabilities (11 moderate, 4 high)
  run `npm audit fix` to fix them, or `npm audit` for details**

Work on a RPI4

Thanks

Stadicus, 2021-12-08:
Try resetting the local repository before checking out the new version:

```sh
$ git reset --hard HEAD
$ git checkout v3.3.0
```

Stadicus, 2021-12-08:
I found another issue with the upgrade process: the command `git describe --tags --abbrev=0` does not return the latest tag because tag `v3.3.0` has no tag date. So if sorted, `v3.2.0` is still the newest tag... Never seen this before.

```sh
$ git for-each-ref --sort=-taggerdate --format '%(refname) %(taggerdate)' refs/tags

refs/tags/v3.2.0 Tue Aug 10 13:23:04 2021 -0400
refs/tags/v3.1.1 Tue Apr 20 19:00:30 2021 -0400
refs/tags/v3.1.0 Wed Apr 14 13:23:59 2021 -0400
refs/tags/v3.0.0 Thu Apr 8 11:47:27 2021 -0400
refs/tags/v2.2.0 Fri Jan 22 14:15:57 2021 -0500
refs/tags/v2.1.0 Tue Dec 15 14:32:46 2020 -0500
refs/tags/v2.0.2 Fri Jul 3 19:43:51 2020 -0400
refs/tags/v2.0.1 Thu May 28 19:03:35 2020 -0400
refs/tags/v2.0.0 Wed Mar 25 16:52:41 2020 -0400
refs/tags/v1.1.9 Sun Feb 23 17:51:57 2020 -0500
refs/tags/v1.1.8 Thu Jan 9 14:20:49 2020 -0500
refs/tags/v1.1.5 Mon Dec 23 00:18:43 2019 -0500
refs/tags/v1.1.4 Wed Dec 4 13:15:37 2019 -0500
refs/tags/v1.1.3 Tue Dec 3 14:43:35 2019 -0500
refs/tags/v1.1.2 Thu Oct 17 13:21:51 2019 -0400
refs/tags/v1.1.1 Tue Oct 1 11:55:45 2019 -0400
refs/tags/v1.1.0 Mon Sep 30 23:02:14 2019 -0400
refs/tags/v1.0.3 Sat Apr 27 16:49:09 2019 -0400
refs/tags/v1.0.2 Tue Mar 12 22:26:11 2019 -0400
refs/tags/v1.0.1 Tue Mar 12 21:59:19 2019 -0400
refs/tags/v1.0.0 
refs/tags/v3.3.0 
```

Ubuntu51, 2021-12-09:
will try again in the evening, but if ed is not with them it will not go with me either.
sorry I do them work

Ubuntu51, 2021-12-09:
The update is through but have had many errors,

**git reset --hard HEAD**

**npm install**
npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but npm-shrinkwrap.json was generated for lockfileVersion@2. I'll try to do my best with it!
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

added 25 packages from 25 contributors, removed 69 packages, updated 59 packages and audited 439 packages in 15.147s

45 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

VajraOfIndra, 2021-12-09:
> Is it working as expected? NPM can give a lot of scary warnings, while still being totally fine. I'm no web dev, it's hard to judge. I updated exactly to the guide without any issues or warnings.

I got similar warnings, the update went well anyway.

Ubuntu51, 2021-12-10:
All right thanks, if the error messages are not a problem all is clear, the update went through and the explorer is running.

Stadicus, 2021-12-10:
Great to hear! :tada:

closed 2021-12-10 by Stadicus