raspibolt/raspibolt #834: Address already in use Electrs server

heredia21, 2021-12-10:
### Describe the issue
I followed the instructions for setting up Electrum server but receiving an error.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/electrs.html)

```
 electrs --conf /data/electrs/electrs.conf
Starting electrs 0.9.3 on aarch64 linux with Config { network: Bitcoin, db_path: "/data/electrs/db/bitcoin", daemon_dir: "/home/bitcoin/.bitcoin", daemon_auth: CookieFile("/home/bitcoin/.bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: Some(1000), reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.3 (Electrum Rust Server)!", args: [] }
[2021-12-10T16:42:15.413Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Error: electrs failed

Caused by:thread 'metrics' panicked at '
    called `Result::unwrap()` on an `Err` value: Os { code: 98, kind: AddrInUse, message: "Address already in use" }Address already in use',  (os error src/metrics.rs98:)30
:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

### Environment
 - Hardware platform: [Raspberry Pi 4
Description:	Debian GNU/Linux 11 (bullseye)

Stadicus, 2021-12-10:
Can you provide the output of this command?

```sh
$ sudo netstat -tulpn | grep LISTEN
```

For me it's the following, with Electrs listening on both ports 50001 (internal) and 50002 (external):

```
admin@raspibolt3:~ $ sudo netstat -tulpn | grep LISTEN
tcp        0      0 127.0.0.1:8333          0.0.0.0:*               LISTEN      39959/bitcoind      
tcp        0      0 127.0.0.1:28333         0.0.0.0:*               LISTEN      39959/bitcoind      
tcp        0      0 127.0.0.1:8334          0.0.0.0:*               LISTEN      39959/bitcoind      
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      54869/lnd           
tcp        0      0 127.0.0.1:50001         0.0.0.0:*               LISTEN      39982/electrs       
tcp        0      0 0.0.0.0:50002           0.0.0.0:*               LISTEN      676/nginx: master p 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      651/sshd: /usr/sbin 
tcp        0      0 127.0.0.1:10009         0.0.0.0:*               LISTEN      54869/lnd           
tcp        0      0 127.0.0.1:3002          0.0.0.0:*               LISTEN      53457/node          
tcp        0      0 127.0.0.1:9050          0.0.0.0:*               LISTEN      654/tor             
tcp        0      0 127.0.0.1:9051          0.0.0.0:*               LISTEN      654/tor             
tcp        0      0 127.0.0.1:4224          0.0.0.0:*               LISTEN      39982/electrs       
tcp        0      0 0.0.0.0:4000            0.0.0.0:*               LISTEN      676/nginx: master p 
tcp        0      0 0.0.0.0:4001            0.0.0.0:*               LISTEN      676/nginx: master p 
tcp        0      0 127.0.0.1:9735          0.0.0.0:*               LISTEN      54869/lnd           
tcp        0      0 127.0.0.1:28332         0.0.0.0:*               LISTEN      39959/bitcoind      
tcp        0      0 127.0.0.1:8332          0.0.0.0:*               LISTEN      39959/bitcoind      
tcp6       0      0 :::22                   :::*                    LISTEN      651/sshd: /usr/sbin 
tcp6       0      0 :::3000                 :::*                    LISTEN      679/node            
tcp6       0      0 ::1:8332                :::*                    LISTEN      39959/bitcoind
```

If you did not start Electrs yet, these ports should not be listed.

heredia21, 2021-12-12:
I believe this is due to having some remaining files on the drive from my Umbrel node. Erasing drive and starting over. Will follow up if there are issues. Thanks for the help.

closed 2021-12-12 by heredia21