raspibolt/raspibolt #840: [Discussion] Installing Mempool: need some help

VajraOfIndra, 2021-12-14:
I've tried to install Mempool on the node following the [repo guidelines](https://github.com/mempool/mempool#manual-installation) and some help from Wiz on their Matrix group (as their installation tips in the repo need a couple of tweaks to avoid running into issues)

I thought I'd share a draft guide of the install so far in case we could finish building it together or else if someone wants to reuse this first part and finish it themselves, that might save you some time.

The draft guide WIP can be found here: [https://github.com/VajraOfIndra/RaspiBolt/blob/mempool/bonus/bitcoin/mempool.md](https://github.com/VajraOfIndra/RaspiBolt/blob/mempool/bonus/bitcoin/mempool.md)

### NGINX

I believe the backend and frontend are properly installed but I'm not sure how to handle the last section about NGINX configuration and certbot: [https://github.com/mempool/mempool#nginx--certbot](https://github.com/mempool/mempool#nginx--certbot)

I assume we have to integrate their nginx settings to our existing nginx.conf file. How should this be done?

I also have a few additional questions:

### Updating NodeJS

They recommend NodeJS v16+. RaspiBolt uses v14 for now. There are a few different methods to update NodeJS, which one would you recommend to update from v14 to v16?

### MariaDB

You MariaDB be run by the root user as recommended in some guides or using admin and sudo is fine in our case?

### Bitcoin RPC username/password

I've used 'raspibolt' and Password [B] for now. Should we instead create a new combination (e.g. 'mempool' and Password [F]) dedicated to Mempool and generate a new 'rpcauth=' line for bitcoin.conf? (I don't understand yet the benefit of creating a new rpcauth, is there one in our case?)

Thanks in advance!

Stadicus, 2021-12-14:
I'm trying to test the setup, but I'm stuck...

First, we should never use the master branch of a repository, which might be unstable. Always check out the latest stable release tag. Here for example:

```sh
git clone https://github.com/mempool/mempool
curl -s https://api.github.com/repos/mempool/mempool/releases/latest | jq -r .tag_name
git checkout v2.2.2
```

I can compile the backend and it's running fine. But then, when compiling the frontend, I don't get the `dist` directory. Your guide derivates a bit from the official instructions here: https://github.com/mempool/mempool#manual-installation

For the frontend, it mentions the additional command `npm run build`, which I think would create the `dist` directory. But I get an error, and no directory. Note: I also copied the `mempool-frontend-config.sample.json` first, without changes.

```
$ cp mempool-frontend-config.sample.json mempool-frontend-config.json
$ npm run build

> mempool-frontend@2.2.2 build
> npm run generate-config && ng build --prod --localize && npm run sync-assets && npm run build-mempool.js

> mempool-frontend@2.2.2 generate-config
> node generate-config.js

mempool-frontend-config.json file found, using provided config
Copied src/index.mempool.html to src/index.html
mempool version 2.2.2
mempool revision 251a1af4
No configuration updates, skipping generated-config.js file update
sh: 1: ng: not found
```

If I get this running, I can work out the NGINX configuraiton. We won't need `certbot`, as this only works for publicly accessible web services with a domain name. We'll stick with our self-signed SSL certificate.

VajraOfIndra, 2021-12-14:
Wdyt regarding 1) MariaDB installation root vs admin user? and 2) the bitcoin RPC username/password, i.e. creating a new pair dedicated to Mempool vs using rapsibolt/PasswordB?

Stadicus, 2021-12-14:
Oh wow, if it's only running from the `master` branch, that's like open-heart surgery still ongoing... I hope they'll get the project into a stable state for the next release. I don't know too much about `npm install --prod`, but usually that skips software development dependencies. Not a good sign if that's not working currently.  We shouldn't release a bonus guide until this is all sorted out.

Now, following your guides including the `npm run build`, it seems to work, thanks!

Stadicus, 2021-12-14:
Regarding NGINX: to expose the web service, we'll just use it as a web server as well, exposing an HTTP service, and then add a similar reverse proxy "external HTTPS --> internal HTTP" on top of that. 

Of course, then we'll also need the systemd units to automatically start front- and backend.

Stadicus, 2021-12-14:
I think the simplest way to update is the following. 

```sh
$ curl -sL https://deb.nodesource.com/setup_16.x | sudo -E bash -
$ sudo apt update
$ sudo apt upgrade
```
But we should check if all other NodeJS applications (BTC RPC Explorer, RTL) also run with NodeJS 16, and then update it for the whole guide. Otherwise, it's going to be messy.

Stadicus, 2021-12-14:
> Wdyt regarding 1) MariaDB installation root vs admin user?

Your installation with user "admin" and `sudo` is fine. There's no functional difference if you run it as user "root". And using "root" is discouraged strongly, some systems even deactivate it completely.

Stadicus, 2021-12-14:
> Wdyt regarding 2) the bitcoin RPC username/password, i.e. creating a new pair dedicated to Mempool vs using rapsibolt/PasswordB?

I don't see a security benefit in creating an additional user, especially as this is the first time this one is used at all. Services from the core guide only use the cookie, maybe that's an option for Mempool as well?

In any case, separate RPC users only make sense if there are multiple applications accessing the backend, potentially running on other systems, managed by different people. You could then monitor and log access, and revoke permissions for a single application if needed, even if you don't have access to all systems. Here, this would feel like total overkill.

VajraOfIndra, 2021-12-14:
> I don't see a security benefit in creating an additional user, especially as this is the first time this one is used at all. 

That was my gut feeling, but wanted to double-check :)

> Services from the core guide only use the cookie, maybe that's an option for Mempool as well?

They said in the chat they don't support cookie auth for now.

> In any case, separate RPC users only make sense if there are multiple applications accessing the backend, potentially running on other systems, managed by different people. You could then monitor [...]

Ah I see, it does make sense.

VajraOfIndra, 2021-12-14:
> Regarding NGINX: to expose the web service, we'll just use it as a web server as well, exposing an HTTP service, and then add a similar reverse proxy "external HTTPS --> internal HTTP" on top of that.

I'm gonna have to draw one of these network diagram to make sense of all this lol :D

RooSoft, 2021-12-16:
> exposing an HTTP service

About that... I've had a converstion with @VajraOfIndra.

I agree that all public traffic should flow through HTTPS no matter what. But my guess is that Raspibolt users might operate locally, or maybe through a VPN.

As far as I know, the Mempool app returns exclusively public information. It's not even possible to log into it with a user. The only sensitive info I see flowing between the user and Mempool may be transaction ids, but they're directly involved in the URL, which is visible nonetheless.

So, with self signed certs, HTTPS encrypts public data but the price to pay is browser warnings. From a user experience perspective, HTTP offers less friction and I don't think it's any less secure than HTTPS here.

Stadicus, 2021-12-17:
> HTTP offers less friction and I don't think it's any less secure than HTTPS here.

Agreed that this is less sensitive than when login information is transmitted. I'd prefer HTTPS more out of principle, also because information could potentially be altered within the network. It's similar to using Electrs over HTTPS, but I agree that the browser warning is a bit inconvenient (at least the first time).

We can offer both methods, though, should be straight-forward.

RooSoft, 2021-12-17:
Chances of data alteration in a local network are usually pretty thin.

I can contribute to this manual as soon as I get my hardware to build a node. As we speak, shipments are being delayed.

Stadicus, 2021-12-17:
> should be straight-forward.

Got mempool.space backend running with the following systemd uning:

```
# RaspiBolt: systemd unit for mempool.space backend
# /etc/systemd/system/mempoolspace-backend.service

[Unit]
Description=mempool.space backend
After=network.target bitcoind.service
After=electrs.service

[Service]
WorkingDirectory=/home/mempool/mempool/backend
#ExecStart=/usr/bin/npm dist/index.js
ExecStart=/usr/bin/npm start
User=mempool

Restart=always
TimeoutSec=120
RestartSec=30

[Install]
WantedBy=multi-user.target
```

Frontend copied over to NGINX directory (fixed typo in current guide):

```sh
$ sudo rsync -av --delete /home/mempool/mempool/frontend/dist/mempool /var/www/
```

The example NGINX config files are not meant to allow for other configurations. I need to dissect them more, but not sure when I'll find the time for this. @RooSoft, your help with this would be much appreciated.

VajraOfIndra, 2022-01-04:
Ok, so I managed to finally make it work in my browser. I updated the draft guide accordingly and created a draft PR #870 

However, I'm not sure my nginx configuration is optimal or secure!

Here are a few questions regarding things to review or improve upon:

* **Preparations section**
  * NodeJS update: for now it points to an external tutorial which list 3 options to upgrade... But updating NodeJS can be very tricky. What is the best way to upgrade for our setup? Later on, it would be great to point to an internal guide tailored to the RaspiBolt setup.

* **Installation section**
  * _Download the repo:_ For now, no verification is done. Is there a way to do _some_ amount of verification of authenticity after we've downloaded the repo and before we install it?
  *  _MariaDB commands:_ The few commands for the MariaDB shell are inside a blockquote. I've used the linguist language symbol "sql" instead of the usual "sh".. is that ok? or should it be something else?

* **nginx section**
  * The nginx config setup is derived from the Raspiblitz mempool.sh install script but adapted to focus only on the secure https connection and by using less snippets. However, I have almost no experiences with the various nginx options within these files and there is therefore a need for a serious review of the various files to ensure the right options and parameters have been used...!
  * In `/etc/nginx/sites-available/mempool-ssl.conf`: for the TLS section within the server context, I reuse the TLS config used for the main guide, is that ok? However, I had to comment out `ssl_session_cache shared:SSL:1m;` as if it was left not commented, an error would result when doing `sudo nginx -t` (see line 324 in the PR)
  * In `/etc/nginx/snippets/nginx-mempool.conf`: I added 2 lines to the default file provided by mempool to restrict connections from the LAN (I added `allow 192.168.0.0/16;` and `deny all;` within the first location context (`location / {...}`) (see line 365-366 in the PR)

* **Update section**
  * I did not test the update section. I'm not sure the process i described is adequate for this program..

* **Uninstall section**
  * Uninstalling MariaDB: See process proposed.. Would it be better to do `sudo apt purge mariadb-server `and `sudo apt purge mariadb-client` instead?

Suggestions for improvements:
* There is no remote Tor connection section yet, this would be a great addition

RooSoft, 2022-02-06:
#870 almost in the books... should we close this #840?

VajraOfIndra, 2022-02-07:
> #870 almost in the books... should we close this #840?

It will be automatically closed once we merge #870 (that's the function of "Fixes #840" in the PR, see [here](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue) for explanations and a list of keywords we can use)

PR #870 (2022-02-07): Bonus guide: Mempool
#### What

This is a proposal to add a bonus guide to install Mempool.

### Why

It is a popular Bitcoin blockchain explorer and mempool visualizer and can be used as an alternative or complement to the BTC RPC Explorer which is installed in the main guide.

Fixes #840 and #651 

#### How

- Bonus guide at /bonus/+bitcoin/mempool.md
- Updated /bonus/+bitcoin/index.md to add a reference to the Mempool guide (under 'Privacy', would another heading be more suitable?)

- The guide requires in-depth review, in particular the nginx configuration section. See Issue #840 for a lis [... cut]
closed 2022-02-07 by VajraOfIndra