raspibolt/raspibolt #843: -bash: /usr/local/bin/lnd: cannot execute binary file: Exec format error

jiyoon-koo, 2021-12-15:
### Describe the issue
I have the bitcoin daemon running for a while, and I decided to install LND, following [this guide](https://raspibolt.org/lnd.html#download). Unfortunately I am running into an error and I'm not sure how it originates / why I'm getting this error. As far as I know, the raspberry pi is running raspbian which should be able to unpack and execute arm64 types.

### Location in guide
https://raspibolt.org/lnd.html#download in subsection `Install LND`

### Screenshots
Added
[image]

### Environment
 - Raspberry Pi 4
 - Raspian GNU 11
 - 32 bit

Thank you for making this guide! It's been really fun so far.

Stadicus, 2021-12-15:
Even if your Raspberry Pi 4 is 64-bit, you still seem to run Raspberry Pi OS 32-bit. Running 64-bit applications won't work unless you set up your Pi with a 64-bit operating system.

But most applications are available for 32 bit architectures, you just need to install the correct release binaries. For LND, this would be the following:

https://github.com/lightningnetwork/lnd/releases/download/v0.14.1-beta/lnd-linux-armv7-v0.14.1-beta.tar.gz

You can also check the (no longer maintained) version 2 of the RaspiBolt guide, which was using a 32-bit operating system:
https://v2.raspibolt.org/

jiyoon-koo, 2021-12-15:
Ah, that makes a lot of sense! When I set up the node (about a month ago), I believe I was using the now "v2" version of raspibolt.org, and finished the Bitcoin section there.

Hopefully, the current guide will still be OK for me to follow along for the Lightning portion of it :) 

Thanks again!

closed 2021-12-15 by jiyoon-koo