raspibolt/raspibolt #847: [BUG] Address already in use

heredia21, 2021-12-17:
Describe the issue

I followed the instructions for setting up Electrum server but receiving an error.
Location in guide

[Link to the corresponding page and section on https://raspibolt.org](https://raspibolt.org/electrs.html)

Error:

```
admin ~ ₿ electrs --conf /data/electrs/electrs.conf
Starting electrs 0.9.3 on aarch64 linux with Config { network: Bitcoin, db_path: "/data/electrs/db/bitcoin", daemon_dir: "/home/bitcoin/.bitcoin", daemon_auth: CookieFile("/home/bitcoin/.bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: Some(1000), reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.3 (Electrum Rust Server)!", args: [] }
[2021-12-17T20:27:19.178Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Error: electrs failed

Caused by:
    Address already in use (os error 98)
thread 'metrics' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 98, kind: AddrInUse, message: "Address already in use" }', src/metrics.rs:30:49
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Environment

    Hardware platform: [Raspberry Pi 4
    Description: Debian GNU/Linux 11 (bullseye)

 sudo netstat -tulpn | grep LISTEN:

```
tcp        0      0 127.0.0.1:50001         0.0.0.0:*               LISTEN      30170/electrs       
tcp        0      0 0.0.0.0:50002           0.0.0.0:*               LISTEN      624/nginx: master p 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      618/sshd: /usr/sbin 
tcp        0      0 127.0.0.1:3002          0.0.0.0:*               LISTEN      30224/node          
tcp        0      0 127.0.0.1:9050          0.0.0.0:*               LISTEN      619/tor             
tcp        0      0 127.0.0.1:9051          0.0.0.0:*               LISTEN      619/tor             
tcp        0      0 127.0.0.1:4224          0.0.0.0:*               LISTEN      30170/electrs       
tcp        0      0 0.0.0.0:4000            0.0.0.0:*               LISTEN      624/nginx: master p 
tcp        0      0 127.0.0.1:28332         0.0.0.0:*               LISTEN      30121/bitcoind      
tcp        0      0 127.0.0.1:8332          0.0.0.0:*               LISTEN      30121/bitcoind      
tcp        0      0 127.0.0.1:8333          0.0.0.0:*               LISTEN      30121/bitcoind      
tcp        0      0 127.0.0.1:28333         0.0.0.0:*               LISTEN      30121/bitcoind      
tcp        0      0 127.0.0.1:8334          0.0.0.0:*               LISTEN      30121/bitcoind      
tcp6       0      0 :::22                   :::*                    LISTEN      618/sshd: /usr/sbin 
tcp6       0      0 ::1:8332                :::*                    LISTEN      30121/bitcoind    
```

Stadicus, 2021-12-17:
Is the `netstat` command run with Electrs running? Because it shows that it's already active and listening on port `50001`. 

Can you `sudo systemctl stop electrs` (and verify that it's stopped with `sudo systemctl status electrs`) and run it again?

heredia21, 2021-12-17:
When I run sudo journalctl -f -u electrs, I get:

```
Dec 17 20:37:58 raspibolt electrs[638]: [2021-12-17T20:37:58.739Z INFO  electrs::daemon] waiting for RPC warmup: Verifying blocks…
Dec 17 20:37:59 raspibolt electrs[638]: [2021-12-17T20:37:59.741Z INFO  electrs::daemon] waiting for RPC warmup: Verifying blocks…
Dec 17 20:38:00 raspibolt electrs[638]: [2021-12-17T20:38:00.742Z INFO  electrs::daemon] waiting for RPC warmup: Verifying blocks…
Dec 17 20:38:01 raspibolt electrs[638]: [2021-12-17T20:38:01.745Z INFO  electrs::daemon] waiting for RPC warmup: Verifying blocks…
Dec 17 20:38:02 raspibolt electrs[638]: [2021-12-17T20:38:02.747Z INFO  electrs::daemon] waiting for RPC warmup: Verifying blocks…
Dec 17 20:38:03 raspibolt electrs[638]: [2021-12-17T20:38:03.783Z INFO  electrs::db] starting config compaction
Dec 17 20:38:03 raspibolt electrs[638]: [2021-12-17T20:38:03.794Z INFO  electrs::db] starting headers compaction
Dec 17 20:38:03 raspibolt electrs[638]: [2021-12-17T20:38:03.794Z INFO  electrs::db] starting txid compaction
Dec 17 20:57:52 raspibolt electrs[638]: [2021-12-17T20:57:52.519Z INFO  electrs::db] starting funding compaction
Dec 17 21:45:28 raspibolt electrs[638]: [2021-12-17T21:45:28.108Z INFO  electrs::db] starting spending compaction

admin ~ ₿ date
Fri Dec 17 22:30:24 GMT 2021
```
Not sure if it is still "syncing" or indexing as I get the error when running the other command, it's been at it for a while after the indexing looks to have completed.. Should I be waiting for something here?

As far as the netstat command with Electrs. How do I check?

heredia21, 2021-12-17:
Ah, it just completed but same error. I stopped the service, verified it stopped then started it again but getting the same error.

Stadicus, 2021-12-19:
The goal is to make sure that Electrs isn't running for sure, and then check if the port is in use. If you stop it, and it still shows up, then it's running already, and the error is caused because you try to start it a second time, with the port already in use.

Can you stop electrs for sure, run `netstat` without the port `50001` showing up, then start Electrs manually?

closed 2022-02-24 by github-actions[bot]