raspibolt/raspibolt #848: [BUG] BTC RPC Explorer breaks on Bitcoin Core restart

Stadicus, 2021-12-19:
### Describe the bug
If Bitcoin Core is restarted, the .cookie file is recreated with new access credentials. Currently, BTC RPC Explorer does not detect this change and just keeps trying to query bitcoind with the old credentials it read from the cookie on startup. This renders the application unusable, until it is restarted (reading the new credentials and accessing bitcoind successfully again).

[image]

### Location in guide
<https://raspibolt.org/btcrpcexplorer.html>

### Expected behavior
BTC RPC Explorer should detect a change in the `.cookie` file content and always use the latest RPC access credentials.

This is not something this guide can fix, so I opened an issue in the project's repository:  
https://github.com/janoside/btc-rpc-explorer/issues/393

VajraOfIndra, 2022-02-20:
Should we create an Issue in the BTC RPC Explorer to ask the author how to deal with this?

kristapsk, 2022-04-20:
I have noticed this too, some time ago.

> This is not something this guide can fix

It could - autorestart BTC RPC Explorer if restart of Bitcoin Core is detected.

> so I opened an issue in the project's repository

But agree that fixing upstream would be the best solution.

kristapsk, 2022-06-27:
Problem still exists, don't think this should have been closed.

Stadicus, 2022-07-11:
I asked in https://github.com/janoside/btc-rpc-explorer/issues/393 if a fix is planned

VajraOfIndra, 2022-09-22:
> It could - autorestart BTC RPC Explorer if restart of Bitcoin Core is detected.

How could this be done?

I've looked at the service file option. I found `BindsTo=` that would allow to automatically stop the Explorer when Bitcoin Core is stopped, but I didn't find anything (so far) that allows to restart it when Bitcoin Core starts... What do you have in mind?

VajraOfIndra, 2022-09-23:
So I looked into the systemd [Unit] options a bit more.

I assume we have the following goals:

1. Start the Explorer if Bitcoin Core starts
2. When starting Bitcoin Core, if the Explorer can't start (e.g. not installed), Bitcoin Core should still starts.
3. The explorer should start only AFTER Bitcoin Core has started
4. Stop the Explorer if Bitcoin Core stops

Also, either:
--5a. If the Explorer is manually started, do not start Bitcoin Core
Or
--5b. If the Explorer is manually started, also start Bitcoin Core

For 1,2,3,4 and 5a, we could use:

`bitcoind.service`:
```
[Unit]
After=network.target
Wants=btcrpcexplorer.service
```

`btcrpcexplorer.service`:
```
[Unit]
After=bitcoind.service
PartsOf=bitcoind.service
```

For 1,2,3,4 and 5b, we would need (`BindsTo` instead of `PartsOf`):

`bitcoind.service`:
```
[Unit]
After=network.target
Wants=btcrpcexplorer.service
```

`btcrpcexplorer.service`:
```
[Unit]
After=bitcoind.service
BindsTo=bitcoind.service
```

---

Wdyt of the above?

- It would solve the issue with BTC RPC Explorer not reading the new cookie file
- It would save some resources by automatically shutting down the Explorer when Bitcoin Core is not running
- It would simplify the shutting down of the node by simply having to terminate Bitcoin Core

Note: The same parametring would benefit Electrs.

kristapsk, 2022-10-13:
One idea - could use `-startupnotify` of `bitcoind`, like we already do with setting permissions for RPC auth cookie (btw, probably will not be needed at some point anymore, if / when https://github.com/bitcoin/bitcoin/pull/26088 gets merged into Bitcoin Core), to `touch` some file somewhere (on `/tmp/`?) , and then use `inotifywait` to restart one or more other services, if they are present (see LND static channel backup script as an example for that).

kristapsk, 2022-10-13:
Or we could even just watch `/home/bitcoin/.bitcoin/.cookie`, which is re-created on each `bitcoind` restart.

VajraOfIndra, 2022-10-20:
> One idea [...]

Wdyt of the proposal above though? Seems simpler and feed two birds with one scone:
1. Work around the BTC RPC Explorer mishandling of the `.cookie` file
2. Save resources by stopping/starting the Explorer when Bitcoin Core stop/starts

PR #1151 (2022-11-06): Fixes #848: Restarting Bitcoin Core without breaking BTC RPC Explorer
#### What/WHy

_(following up the clossure of PR #1118 - this PR has a more focused scope)_

tl;dr: A PR to fix #848 : Restarting Bitcoin Core does not break BTC RPC Explorer

Bug: When Bitcoin Core is restarted, a new cookie file is generatd. But the Explorer does not notice and still use the older cookie file, which prevents the user to acces the Explorer (due to wrong Bitcoin Core RPC authentification). This is a mishandling of the cookie file by the Explorer and should be fixed in the Explorer codebase ittself. 

However [... cut]
commit 3ba5312a (2022-11-06): Fixes #848: Restarting Bitcoin Core without breaking BTC RPC Explorer (#1151)

* Added PartOf bitcoind in service file

To shut down the Explorer when Bitcoind shuts down

* Adds Wants btcrpcexplorer to bitcoind service

Allows for the Explorer to start when Bitcoin Core starts/restarts

* Removes network.target

Not required as at boot time, the Explorer is started after bitcoind which starts after network.target.
closed 2022-11-06 by VajraOfIndra