raspibolt/raspibolt #849: Serve navigation to available webservices directly on https://raspibolt.local

Stadicus, 2021-12-19:
### What & why

As a RaspiBolt user, I'd like to enter `raspibolt.local` in my browser, and get an overview of all available web services, so that I don't need to memorize all these specific ports like `raspibolt.local:4000`.

### How

I doubt that a proper homepage for the RaspiBolt makes sense (too little content, just a few links), but I could imagine a top- or side-bar using frames could be a neat addition:

* Users navigates to https://raspibolt.local
* They see 
  * a left sidebar (with a logo, listing links to Bitcoin Explorer, RTL, etc.)
  * a main section showing the Bitcoin Explorer by default

What's needed?
* a static HTML page defining the [frameset](https://www.w3schools.in/html-tutorial/frames/)
* a static HTML page for the sidebar (or topbar), with a neat logo and some links
* served over NGINX with TLS on standard HTTPS port 443 using a self-signed certificate ([see tutorial](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-nginx-in-ubuntu-16-04))

### Additional context

Maybe a cool feature to implement by a new contributor?

VajraOfIndra, 2021-12-20:
That sounds really really cool! :) I want it now! ;)

If the system overview script could become part of the main guide (big if, but see comment in #850 ), would it be possible to have the overview as the welcome page? (and then sidebar links for the various webUIs: Explorer and RTL for now (maybe also Lightning Terminal UI & Mempool later on..))

Stadicus, 2021-12-22:
> would it be possible to have the overview as the welcome page

Neat idea, but probably a completely separate codebase. I guess it either needs to be rendered interactively by a server-side script (e.g. Python), but that could open the system to all kinds of vulnerability if not done carefully. Or we have a service that queries the data from time to time and renders a static website that can be served by NGINX.

VajraOfIndra, 2021-12-22:
> Or we have a service that queries the data from time to time and renders a static website that can be served by NGINX.

That sounds great, better that no system overview at all.

Stadicus, 2021-12-23:
I guess the web-version of the system overview can be independent of the general navigation proposed here, as to not block it.

In any case: we should do it as nerdy / command-line like as possible :-P

VajraOfIndra, 2021-12-23:
> I guess the web-version of the system overview can be independent of the general navigation proposed here, as to not block it.

Absolutely, this could be somehting that is added later if we think it still makes sense.

VajraOfIndra, 2022-02-17:
I posted this proposal in the Telegram group to see if someone would be motivated to implement it.

openoms, 2022-02-17:
This project might be a good fit:
https://github.com/bastienwirtz/homer
proposed for the raspiblitz: https://github.com/rootzoll/raspiblitz/pull/2541/files

VajraOfIndra, 2022-02-17:
> This project might be a good fit: https://github.com/bastienwirtz/homer proposed for the raspiblitz: https://github.com/rootzoll/raspiblitz/pull/2541/files

Interesting, thanks, I'll play with this over the weekend!

Stadicus, 2022-02-18:
The Homer project mentioned by Openoms might be a good fit, especially due to the simple YAML configuration.

VajraOfIndra, 2022-02-19:
So I played around with the Homer dashboard on my node, it's pretty cool! :)

I created a "homer" branch in the RaspiBolt repo: https://github.com/raspibolt/raspibolt/tree/homer. It contains a first draft of the guide. But this is just to get the ball rolling, feel free to change whatever you want inside!

A live version of the draft can be seen here based on a branch in my repo: https://vajraofindra.github.io/RaspiBolt/

So far, I simply added the BTC RPC Explorer and RTL services:
[image]

--- 

Here are a few thoughts and questions that we could discuss:

### Structure

* IMO, I like the idea of creating a new first-level chapter after "Bitcoin" and "Lightning". I entitled it "Node Management", but there might be much better names. Inside that section, the guide is entitled "Web dashboard" (again, we could discuss this title).

[image]

* This section could contain additional guides in the future, such as the raspibolt/motd script (as has been discussed before).
* What do you think of also installing a SCB monitoring app that could be added to the dashbaord: e.g. [RPi-Monitor](https://github.com/XavierBerger/RPi-Monitor) or other similar sysadmin programs?

Otherwise, would you prefer to have it as a bonus guide instead? or else in the core guide but within another structure?

---

### Design

* Logos: I tried to use the app logos but Homer couldn't display my files. I put them in the correct folder (/dist/assets/tools), as .png and with the correct linls in the yaml config file.. but I must be missing something to make them work. Any ideas?
* Icons: Icons mostly work and there are a huge list at https://fontawesome.com/icons. But sometimes some icons wouldn't be rendered om the webpage.. not entirely sure why.
* Colours: It'd be awesome to change the colors to use the RaspiBolt colours, but I didn't find where the banner colour is defined (does not seem to be in the config file..)

---

### Top banner links

There is a useful banner at the top that can contains links to external websites. So far, I added three links:
* Guide: https://raspibolt.org/
* Community: https://t.me/raspibolt/
* Contribute: https://github.com/raspibolt/raspibolt

Wdyt? any ideas for additional links? or changes to these three links?

---

### Top message

At the top of the main screen there is an optional message box. For now I just put a simple title, icon and one-liner message. However, the box can contain links and even get text from an API call. 

Any ideas what would be cool to put there?

---

### Services

So far in the core guide we have only 2 web services: BTC RPC Explorer and RTL

* We could a 'Homer' section in the bonus guides that includes a web service to add them to the dashboard: i.e. for now Mempool, and soon once the PRs are merged, ThunderHub, Lightning Terminal and Terminal Web

* I like the idea of adding links to useful external websites in the dashboard, e.g.:
  * in the Bitcoin section:
    * [Mempool](https://mempool.space/): for fee estimation and mempool visualization
  * in the Lightning section:
    * [Amboss](https://www.amboss.space/node/02acd93e3352fd59066ca3f23e8865de1926301e8be03c6a52f0f7e43533fe9888): a link to the node information page in Amboss: 
    * [Terminal Web](https://terminal.lightning.engineering/#/02acd93e3352fd59066ca3f23e8865de1926301e8be03c6a52f0f7e43533fe9888): a linlk to the node information page in Terminal Web

I think this would make  the dashboard even more useful, however, this could leak the IP address of the user, correct? So maybe with a warning about it? Or would you rather not have links to external services like this?

---

### Sysadmin

Above I asked the question about should we install a program like RPi-Monitor to get a web-based node monitoring tool.  

I also saw dashboard with server stats as part of the message bos, e.g.:
[image]
But I have no idea how this could be done.. would that be something we'd like to add? if so, how?

---

Again, this is just to get the ball rolling, feel free to change anything you want in the branch.. structure, file names and content! :)

VajraOfIndra, 2022-02-22:
Assuming we want to add a web services dashboard to the main guide (vs having it as a bonus guide), another possible flow/structure could be:

1. In the "Rasperry Pi" chapter of the guide, we'd install a lightweight sysadmin web service, *e.g.* Cockpit or an alternative, after "Privacy" and under a sub-heading such as "System overview" or "Sysadmin":
[image]

2. Then we move on to installing a lightweight web services dashboard, *e.g.* Homer or an alternative, under a sub-heading such as "Web dashboard".
[image]

At this stage of the guide, the dashboard would contain only the sysadmin web service + all the toolbard links. But the config file would contain commented out lines for the other services that will be unlocked later in the guides:
[image]

3. Then, in the BTC RPC Explorer and Ride The Lighnting guides, we add a "Homer" section towards the end where we simply ask to uncomment the relevant lines in the config file: 

*E.g.* BTC RPC Explorer:
[image]

*E.g.* Ride the Lightning:
[image]

4. We would add a similar sections to the bonus guides that contain a self-hosted web service.

---

IMO, both structures could work quite well, although this one is a bit more conservative (it preserves the first-level headings).

Wdyat? :)

(*Note:* if a sysadmin web service is judged not desired for the guide, the following flow could still be done but without the "Cockpit" installation and the dashboard would simply contain the Bitcoin and Lightning web service groups.

This structure can temporarily be seen on my local GitHub Pages: https://vajraofindra.github.io/RaspiBolt/

RooSoft, 2022-02-23:
That's a good idea, but why not explain how to add these homer config lines directly in the related sections?

Cause with those commented lines would end up making no sense on a node that are not intended to run those services.

VajraOfIndra, 2022-02-23:
> That's a good idea, but why not explain how to add these homer config lines directly in the related sections?
> 
> Cause with those commented lines would end up making no sense on a node that are not intended to run those services.

Yes, this would work as well. I just thought that explaining _where_ to paste new line was less easy to explain and more error-prone that asking to uncomment a section already in the config file... But I do agree that it's not ideal to have extra config lines, albeit as comments, that won't necessarily be used by everyone. 
So I'd be appy to switch to pasting new lines instead. We could use line numbers to make it clear where the added lines have to be copied, and use  the `--linenumbers` flag to open `nano` with numbered lines.

RooSoft, 2022-02-24:
Line numbers probably won't do it if this file is different from one user to another...

VajraOfIndra, 2022-02-28:
> Line numbers probably won't do it if this file is different from one user to another...

I suppose for the default setup in a guide, line numbers could be useful for when the users go through the guide the first time. Then they might want to do their own thing with teh config file.. but they'll know that the line numbers won't be mtching their file anymore.

But we could go without line numbers, it should still be fairly simple to explain where what goes where.

VajraOfIndra, 2022-03-04:
After some more thoughts on the topic:
* Cockpit is probably NOT a good fit with the core guide: being a GUI, it incentivizes users to use it rather than the command line for sysadmin operations -> less didactic
* the installation could be done as for nginx and Tor, i.e. initial set up at the start and adding config as needed. In our case, a three-steps installation: 1) installation and basic configuration in the "Raspberry Pi" chapter; 2) adding the web service config lines after installing the Explorer; and 3) same after installing RTL 

My proposal would be:
* Install Homer after the "Privacy" section in the "Raspberry Pi" chapter, under a generic heading (e.g. "Web dashboard")
  * Explains that if a user does not plan to use self-hosted web services they can skip this section
* The initial config would contain all the useful external links and logos and web services categories but no links to any self-hosted services -> https://github.com/VajraOfIndra/RaspiBolt/blob/homer-new/guide/raspberry-pi/web-dashboard.md
* We would then have "Homer" sections within any subsequent guides that provide a web service:
  * _In the core guide that would be:_
    * the BTC RPC Explorer guide -> https://github.com/VajraOfIndra/RaspiBolt/blob/homer-new/guide/bitcoin/blockchain-explorer.md#homer
    * the RTL guide -> https://github.com/VajraOfIndra/RaspiBolt/blob/homer-new/guide/lightning/web-app.md#homer
  * _In the bonus section, that would be an optional section in:_
    * Mempool -> https://github.com/VajraOfIndra/RaspiBolt/blob/homer-new/guide/bonus/bitcoin/mempool.md#optional-homer
    * the white paper download -> https://github.com/VajraOfIndra/RaspiBolt/blob/homer-new/guide/bonus/bitcoin/white-paper.md#optional-homer
    * and later once more PRs are merged, Ligthning Terminal, ThunderHub and many others...!

The branch implementing these changes can be viewed live here: https://vajraofindra.github.io/RaspiBolt/

---

**Dashboard content:**
* Header: RaspiBolt title and logo from an imported PNG picture 
* Top banner:
  * Link to RaspiBolt guide
  * Links to all the RaspiBolt community groups: GitHub, Telegram, Twitter, Reddit, Amboss
  * Links to the user's node profile on various LN explorer: Amboss, Web Terminal, LnRouter Debugger, LNnodeinsights, LN+ and 1ML
 * A message fetched from an API: for now the next block suggested miner fee from mempool.space (but could be another API, if possible self-hosted, like the BTC RPC Explorer random bitcoin quote if we can make it work; or something else?)
 * The web services organised by categories. In the core guide, simply "Bitcoin" and "Lightning". 
   * All web services would have their logos from an imported PNG

Final version of the dashboard at the end of the core guide would be something like this:
[image]

I don't know how to merge my changes to the Homer branch in the raspibolt repo as there are conflicts. I could either create a PR or else let me know how to force push my changes to the branch.

--- 
Some remaining questions:
* The dashboard is served on port 4091. Can we, and do we want to, set it up so that we don't have to type any ports and simple go to `https://raspibolt.local` as initially proposed by @Stadicus in the OP? 
* Is Homer the right choice for our goals or should we consider something else entirely..? (in that case we could transform this Homer dashboard guide as a bonus guide)

Looking forward to hear your thoughts and suggestions on all this.

VajraOfIndra, 2022-03-12:
@Stadicus @RooSoft What's your thoughts on the above? and on the last two questions?

Is it something you'd like in the main guide? 
If yes, I'll make a PR for you to review. If not, I could transform it in a bonus guide.

closed 2022-05-18 by github-actions[bot]