raspibolt/raspibolt #853: Permissions issue bitcoind.service

c-mannix, 2021-12-24:
### Describe the issue
Followed the guide up to installing Bitcoin Core and my node is currently synching. However, when I check the status with `systemctl status bitcoind.service` I see `Warning: some journal files were not opened due to insufficient permissions.`  when running the command as `admin`.
 I get a different message when running the command as the `bitcoin` user, 
 `Warning: journal has been rotated since unit was started and some journal files were not opened due to insufficient permissions, output may be incomplete.`
 Do I need to change some permissions?

### Location in guide
https://raspibolt.org/bitcoin-core.html#running-bitcoind

### Screenshots
Screenshot with `admin` user:
[image]
Screenshot with `bitcoin` user:
[image]

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

Stadicus, 2021-12-26:
I think that can be safely ignored for bitcoind, as we query the `debug.log` file directly, not using system logging.

If you run the "status" command you referenced with `sudo`, you should get the full output (but not much, or not more than in the log file)

c-mannix, 2021-12-26:
It threw me off because the image shown in the guide did not have the warning that I was seeing so I thought I might have set some permissions incorrectly.

Stadicus, 2021-12-27:
Good point. I just checked, and I get the same warning myself. The command listed in the guide is perfectly fine, we should just remove the following lines:

https://github.com/raspibolt/raspibolt/blob/3349f4d508b39121ca68303e75853efd49cecd8e/bitcoin-core.md?plain=1#L339-L342

c-mannix, 2021-12-27:
May be a good idea to add the warning message as well. If someone doesn't see that they still may think they did something wrong because it differs from the actual output they see when running the command.

PR #855 (2021-12-28): Update bitcoind.service status output
Update bitcoin.service status output to help avoid confusion.

Closes #853
closed 2021-12-28 by Stadicus