raspibolt/raspibolt #857: [Discussion] installing c-lightning: need some help

real-doman, 2021-12-27:
### What & why
Add installation of c-lightning to give have more choises on which LN implementation you want to use on your raspibolt
I don't know if this should be a part of the bonus section or if it should be put elsewhere

I am far from being a system expert, nevertheless I started the guide.

The current guide is still in progress, because I run into an issue while the initial setup. You can find the draft guide [here](https://github.com/real-doman/RaspiBolt/blob/c-lightning/c-lightning.md)

I have tried to install c-lightning on the raspibolt, but unfortunately I don't get it running properly. To get an idea of the installation process I used the [raspiblitz c-lightning installation file](https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/cl.install.sh)

The installation worked well for me, but I have troubles with the `config` file. When I try to run c-lightning for the first time I get a permisson error:
```
$ lightningd --conf=/data/lightningd/config
> lightning_connectd: Could not open Tor cookie file '/run/tor/control.authcookie': Permission denied (version 0.10.2)
> 0x557b311ae3 send_backtrace
        common/daemon.c:33
> 0x557b317e73 status_failed
        common/status.c:209
> 0x557b30db03 negotiate_auth
        connectd/tor_autoservice.c:260
> 0x557b30e2e7 tor_fixed_service
        connectd/tor_autoservice.c:350
> 0x557b308d7f setup_listeners
        connectd/connectd.c:1427
> 0x557b309017 connect_init
        connectd/connectd.c:1512
> 0x557b30ab4f recv_req
        connectd/connectd.c:1921
> 0x557b311c73 handle_read
        common/daemon_conn.c:30
> 0x557b348903 next_plan
        ccan/ccan/io/io.c:59
> 0x557b348edb do_plan
        ccan/ccan/io/io.c:407
> 0x557b348f9b io_ready
        ccan/ccan/io/io.c:417
> 0x557b34ae9b io_loop
        ccan/ccan/io/poll.c:453
> 0x557b30b1f3 main
        connectd/connectd.c:2006
> 0x7fbdd64217 ???
        ???:0
> 0x557b305647 ???
        ???:0
>  ightningd: connectd failed (exit status 242), exiting.
```
My `cl` user is part of the `debian-tor` group and should have read permisson. So I don't know why the permisson is denied :(

VajraOfIndra, 2021-12-29:
Great initiative, it'd be awesome to have a guide for c-lightning (and later related add-ons) somewhere in the guide. The bonus section seems the easiest choice for now. If it were to become part of the main guide we would probably have to think about to deal with the split between LND and c-lightning... dual-install? (probably unnecessary for most users), either-or choice (nice but requires a very clear presentation to avoid risks of confusion for new users..). Anyways, just my 2 sats, looking forward to hear what others have to say! :)

> The current guide is still in progress, because I run into an issue while the initial setup. You can find the draft guide here

Nice! I'll give it a try in the next few days/weeks.

> The installation worked well for me, but I have troubles with the config file. When I try to run c-lightning for the first time I get a permisson error:

Could it be because you run `lightningd` with the "admin" user? (based on the end of your guide). The "admin" user is not part of the "debian-tor" group so wouldn't be able to read the cookie. Do you get the same error if you run `lightningd` with the cl user?

openoms, 2022-01-05:
I run the ligthningd service and keep plugins, config etc. under the bitcoin user, the same running lnd and bitcoind.
CL needs access to bitcoin-cli and the Tor cookie as well.

Check in your `/etc/tor/torrc` if the lines:
```
CookieAuthentication 1
CookieAuthFileGroupReadable 1
```
are present.

openoms, 2022-02-21:
@real-doman did you get c-lightning working yet? Would be really good to see a guide on this, would be happy to contribute.

real-doman, 2022-02-22:
@openoms no. I gave the installation a new try and updated the guide a bit. 
To get an idea about all of this I am using the c-lightning install script from raspiblitz. At [that part](https://github.com/rootzoll/raspiblitz/blob/320015dc6086fb2bf238527d67581eaa574ac638/home.admin/config.scripts/cl.install.sh#L37) an error is thrown. Can I create this file manually somehow or why is the file even missing?

openoms, 2022-02-22:
unfortunately the error you ran into is because you are installing from the zip file. This will be sorted for the next CL release, but until then install from source: https://github.com/rootzoll/raspiblitz/pull/2936
I will probably stick to installing from the source anyway as it is easier to update to the latest commits or PRs from it.

real-doman, 2022-02-24:
Thats sad, because I am more familiar with this way. I can stick to the current lnd installation guide more easily. How ever, I tried to clone the repository now, unfortunately the error still appears.

real-doman, 2022-02-25:
I was able to fix my issues and afaik the installation of c-lightning was successful.
My guide would continue with the first manual start of c-lightning, but I am not familiar with all of the commands, so I would need help for this part.

For now the verification process is still missing in my guide. that needs to be added aswell.

kendricng, 2022-03-08:
@real-doman It seems like I am getting some permission errors when trying to run `lightningd`. See below:

```
cl@raspibolt:~ $ lightningd --conf=/data/lightning/config
error starting plugin '/home/cl/.lightning/plugins/bcli': opening pipe: Permission denied
```

How can I trace where the permissions are breaking down? It seems like I'm not able to open the config file. I tried changing the file ownership using `sudo chown` but that also didn't work.

```
admin@raspibolt:~ $ ls -al /data/lightning/plugins
-rwxr-xr-x  1 cl cl 3062192 Mar  8 01:52 bcli

admin@raspibolt:~ $ ls -al /data/lightning
-rw-r--r--  1 cl      cl         227 Mar  8 02:12 config
```

openoms, 2022-03-08:
@kendricng 
run `ls -al /data/lightning/plugins` with the `cl` user to test. Fix the permissions of the upstream directories also if there is an issue.
`bcli` needs `bitcoin-cli` afaik so make sure the `cl` user can start bitcoin-cli as well. Will probably need to symlink the bitcoin direcroty to the default path in `cl` to appear as:
`/home/cl/.bitcoin`
the bitcoin directory also needs to be group readable and `cl` tp be part of the group `bitcoin`.

In the raspiblitz we kept C-lightning under the `bitcoin` user to prevent these issues.

kendricng, 2022-03-10:
This is what I found when running `ls -al /data/lightning`:

```
cl@raspibolt:~ $ ls -al /data/lightning
total 2092
drwxr-xr-x 25 cl cl   4096 Mar 10 14:36 .
drwxr-xr-x  8 cl cl   4096 Mar  8 02:01 ..
-rw-r--r--  1 cl cl    161 Mar  8 01:43 action.yml
drwxr-xr-x  3 cl cl   4096 Mar  8 02:03 bitcoin
drwxr-xr-x  5 cl cl   4096 Mar  8 01:44 ccan
-rw-r--r--  1 cl cl   2512 Mar  8 01:44 ccan-asort.o
-rw-r--r--  1 cl cl  21808 Mar  8 01:44 ccan-bitmap.o
-rw-r--r--  1 cl cl   2472 Mar  8 01:44 ccan-bitops.o
-rw-r--r--  1 cl cl   8992 Mar  8 01:44 ccan-breakpoint.o
-rw-r--r--  1 cl cl  64072 Mar  8 01:44 ccan-cdump.o
-rw-r--r--  1 cl cl  26280 Mar  8 01:44 ccan-closefrom.o
-rw-r--r--  1 cl cl   1002 Mar  8 01:43 ccan_compat.h
-rw-r--r--  1 cl cl  10480 Mar  8 01:44 ccan-crc32c.o
-rw-r--r--  1 cl cl   9672 Mar  8 01:44 ccan-crypto-hkdf.o
-rw-r--r--  1 cl cl  11912 Mar  8 01:44 ccan-crypto-hmac.o
-rw-r--r--  1 cl cl 506152 Mar  8 01:44 ccan-crypto-ripemd160.o
-rw-r--r--  1 cl cl  51272 Mar  8 01:44 ccan-crypto-sha256.o
-rw-r--r--  1 cl cl  14888 Mar  8 01:44 ccan-crypto-shachain.o
-rw-r--r--  1 cl cl  23224 Mar  8 01:44 ccan-crypto-siphash24.o
-rw-r--r--  1 cl cl   2000 Mar  8 01:44 ccan-err.o
-rw-r--r--  1 cl cl   8056 Mar  8 01:44 ccan-fdpass.o
-rw-r--r--  1 cl cl  42040 Mar  8 01:44 ccan-htable.o
-rw-r--r--  1 cl cl   6640 Mar  8 01:44 ccan-ilog.o
-rw-r--r--  1 cl cl  30856 Mar  8 01:44 ccan-intmap.o
-rw-r--r--  1 cl cl   9592 Mar  8 01:44 ccan-io-fdpass.o
-rw-r--r--  1 cl cl  60368 Mar  8 01:44 ccan-io-io.o
-rw-r--r--  1 cl cl  45880 Mar  8 01:44 ccan-io-poll.o
-rw-r--r--  1 cl cl  24808 Mar  8 01:44 ccan-isaac64.o
-rw-r--r--  1 cl cl  24808 Mar  8 01:44 ccan-isaac.o
-rw-r--r--  1 cl cl  17768 Mar  8 01:44 ccan-json_escape.o
-rw-r--r--  1 cl cl  40288 Mar  8 01:44 ccan-json_out.o
-rw-r--r--  1 cl cl   9744 Mar  8 01:44 ccan-list.o
-rw-r--r--  1 cl cl   8728 Mar  8 01:44 ccan-membuf.o
-rw-r--r--  1 cl cl  13480 Mar  8 01:44 ccan-mem.o
-rw-r--r--  1 cl cl   9232 Mar  8 01:44 ccan-noerr.o
-rw-r--r--  1 cl cl  57336 Mar  8 01:44 ccan-opt-helpers.o
-rw-r--r--  1 cl cl  35720 Mar  8 01:44 ccan-opt.o
-rw-r--r--  1 cl cl  16040 Mar  8 01:44 ccan-opt-parse.o
-rw-r--r--  1 cl cl  29776 Mar  8 01:44 ccan-opt-usage.o
-rw-r--r--  1 cl cl  19912 Mar  8 01:44 ccan-pipecmd.o
-rw-r--r--  1 cl cl  32256 Mar  8 01:44 ccan-ptr_valid.o
-rw-r--r--  1 cl cl  25792 Mar  8 01:44 ccan-rbuf.o
-rw-r--r--  1 cl cl   6816 Mar  8 01:44 ccan-read_write_all.o
-rw-r--r--  1 cl cl  16280 Mar  8 01:44 ccan-str-base32.o
-rw-r--r--  1 cl cl   9080 Mar  8 01:44 ccan-str-hex.o
-rw-r--r--  1 cl cl  21144 Mar  8 01:44 ccan-strmap.o
-rw-r--r--  1 cl cl   4696 Mar  8 01:44 ccan-str.o
-rw-r--r--  1 cl cl  11952 Mar  8 01:44 ccan-take.o
-rw-r--r--  1 cl cl  11240 Mar  8 01:44 ccan-tal-grab_file.o
-rw-r--r--  1 cl cl  16520 Mar  8 01:44 ccan-tal-link.o
-rw-r--r--  1 cl cl  80200 Mar  8 01:44 ccan-tal.o
-rw-r--r--  1 cl cl  45312 Mar  8 01:44 ccan-tal-path.o
-rw-r--r--  1 cl cl  34208 Mar  8 01:44 ccan-tal-str.o
-rw-r--r--  1 cl cl  13336 Mar  8 01:44 ccan-time.o
-rw-r--r--  1 cl cl  54296 Mar  8 01:44 ccan-timer.o
-rw-r--r--  1 cl cl   8224 Mar  8 01:44 ccan-utf8.o
-rw-r--r--  1 cl cl 117742 Mar  8 01:43 CHANGELOG.md
drwxr-xr-x  3 cl cl   4096 Mar  8 01:49 channeld
-rw-r--r--  1 cl cl    141 Mar  8 01:43 .clang-format
drwxr-xr-x  3 cl cl   4096 Mar  8 01:51 cli
drwxr-xr-x  2 cl cl   4096 Mar  8 01:49 closingd
drwxr-xr-x  3 cl cl  12288 Mar  8 01:52 common
-rw-r--r--  1 cl cl    227 Mar 10 14:36 config
-rwxr-xr-x  1 cl cl  12891 Mar  8 01:43 configure
-rw-r--r--  1 cl cl   2250 Mar  8 01:44 config.vars
drwxr-xr-x  3 cl cl   4096 Mar  8 01:49 connectd
drwxr-xr-x 13 cl cl   4096 Mar  8 01:43 contrib
drwxr-xr-x  2 cl cl   4096 Mar  8 01:52 devtools
-rw-r--r--  1 cl cl    193 Mar  8 01:43 .dir-locals.el
drwxr-xr-x  4 cl cl  12288 Mar  8 01:57 doc
-rw-r--r--  1 cl cl   4438 Mar  8 01:43 Dockerfile
-rw-r--r--  1 cl cl    316 Mar  8 01:43 .editorconfig
drwxr-xr-x  9 cl cl   4096 Mar  8 01:45 external
drwxr-xr-x  9 cl cl   4096 Mar  8 01:58 .git
-rw-r--r--  1 cl cl    483 Mar  8 01:43 .gitattributes
drwxr-xr-x  4 cl cl   4096 Mar  8 01:43 .github
-rw-r--r--  1 cl cl    946 Mar  8 01:43 .gitignore
-rw-r--r--  1 cl cl   1418 Mar  8 01:43 .gitlab-ci.yml
-rw-r--r--  1 cl cl    665 Mar  8 01:43 .gitmodules
drwxr-xr-x  3 cl cl   4096 Mar  8 01:48 gossipd
-rw-r--r--  1 cl cl   1365 Mar  8 01:47 header_versions_gen.h
drwxr-xr-x  2 cl cl   4096 Mar  8 01:47 hsmd
-rw-r--r--  1 cl cl   1208 Mar  8 01:43 LICENSE
lrwxrwxrwx  1 cl cl     15 Mar  8 02:38 lightning -> /data/lightning
drwxr-xr-x  3 cl cl   4096 Mar  8 01:51 lightningd
-rw-r--r--  1 cl cl  33549 Mar  8 01:43 Makefile
drwxr-xr-x  3 cl cl   4096 Mar  8 01:49 onchaind
drwxr-xr-x  2 cl cl   4096 Mar  8 01:49 openingd
drwxr-xr-x  4 cl cl   4096 Mar  8 01:53 plugins
-rw-r--r--  1 cl cl  11801 Mar  8 01:43 README.md
-rw-r--r--  1 cl cl   3471 Mar  8 01:43 requirements.lock
-rw-r--r--  1 cl cl    260 Mar  8 01:43 requirements.txt
drwxr-xr-x  6 cl cl   4096 Mar  8 01:43 tests
drwxr-xr-x  4 cl cl   4096 Mar  8 01:58 tools
-rw-r--r--  1 cl cl     52 Mar  8 01:47 version_gen.h
drwxr-xr-x  3 cl cl   4096 Mar  8 01:51 wallet
drwxr-xr-x  3 cl cl   4096 Mar  8 01:53 wire
```

Will I need to recursively change access to 755? I think what happened was that I installed `c-lightning` using admin first, `mv` to `/data/lightning` and then did the symlink.

Checked `bitcoin-cli` and the symlinks and I think they're good:

```
cl@raspibolt:~ $ bitcoin-cli --version
Bitcoin Core RPC client version v22.0.0
```

```
cl@raspibolt:~ $ ls -al
lrwxrwxrwx 1 cl   cl     13 Mar  8 02:41 .bitcoin -> /data/bitcoin
lrwxrwxrwx 1 cl   cl     15 Mar  8 02:42 .lightning -> /data/lightning
```

```
cl@raspibolt:~ $ ls -al /data
drwxr-xr-x  5 cl   cl    4096 Mar 10 14:13 bitcoin
drwxr-xr-x 25 cl   cl    4096 Mar 10 14:36 lightning
```

openoms, 2022-03-10:
@kendricng be aware of the `dot` front of `.lightning`

kendricng, 2022-03-10:
@openoms Yup. I'm aware. Also double checking the Raspibolt guide that I followed when I initially installed `bitcoin` and `lnd` also they didn't have the dots in front of them as well.

https://raspibolt.org/guide/lightning/lightning-client.html

openoms, 2022-03-10:
LND has nothing to do with this.
the `lightning` directory has the source code from which the binaries are built.
the `.lightning` contains the C-lightning data like the `hsm_secret` , `[lightning.sqlite3` and config file.

kendricng, 2022-03-10:
I understand. What I mean is that I mirrored the guide from Raspibolt from the instructions there and followed @real-doman 's test guide for `c-lightning` here.

https://github.com/real-doman/RaspiBolt/blob/c-lightning/c-lightning.md

I don't see the `.lightning` directory in this version so am not sure where the C-lightning data would be stored.

kendricng, 2022-03-10:
@real-doman in the Data Directory portion of your guide, instead of moving over the binary files for `cl` to the new file, I would just create a new directory and use that to symlink, a la (from the Raspiblitz shell file):

```
admin@raspibolt:~ $ sudo rm -rf /home/cl/.lightning
admin@raspibolt:~ $ sudo mkdir -p /data/.lightning
admin@raspibolt:~ $ sudo ln -s /data/.lightning /home/cl
admin@raspibolt:~ $ sudo chown -R cl:cl /data/.lightning
admin@raspibolt:~ $ sudo chown -R cl:cl /home/cl
admin@raspibolt:~ $ sudo ln -s /data/.lightning /home/admin/.lightning
```

real-doman, 2022-03-10:
> I propose this change because the `cl` user didn't know how to access the files needed to run `lightningd` and hence I was getting Permission Denied errors.

Agree, that part needs to be changed. Using `mv` command feels a bit strange for me aswell.

But I can't say anything to the permisson denied errors... For me the guide worked if I run everything in exactly same order as in the guide. I don't run into permisson denied error on my raspibolt for some reason.

openoms, 2022-04-04:
did a cleaned script with which you can run line by line and use for reference to install CLN on raspibolt: https://gist.github.com/openoms/ef2d71f98d0f6ee6441e91f840b89e4c
note it was not tested, but should work as quickly changed the user and paths to the raspibolt environment
can also comment below
feel free to use it to start a new guide or correct the one already there

For example simplified the symlinks:
```
next one with the symlinks:
I would do:
# Store the lightning data in /data and symlink to /home/cl/"
sudo rm -rf /home/cl/.lightning # not a symlink, delete
sudo mkdir -p /data/.lightning
sudo ln -s /data/.lightning /home/cl/

# Create a symbolic link, which points from your admin home directory into the lightning directory
sudo ln -s /data/.lightning /home/admin/
```

then there are things like:
* git verification 
* creating an hsm_secret from seed 
* systemd service  
* setting up the backup plugin
* hsm_secret encryption

which are automated in the raspiblitz and can be tackled one by one after the basic installation is working.

openoms, 2022-04-05:
@real-doman I would suggest to open a PR on to the raspibolt repo (can make it a draft until it is ready to be merged) then your guide becomes visible, people can review, comment on the code and will get there soon

openoms, 2022-04-10:
Note the introduction of poetry for the pyhton depandencies and other changes: https://github.com/ElementsProject/lightning/blob/master/doc/INSTALL.md#to-build-on-ubuntu

The install script will need to be refactored, see changes in: https://github.com/rootzoll/raspiblitz/issues/3054

openoms, 2022-05-02:
note https://github.com/raspibolt/raspibolt/pull/1005 fixes this so can be closed

VajraOfIndra, 2022-05-03:
Fixed with #1005, will now close the Issue.

closed 2022-05-03 by VajraOfIndra