raspibolt/raspibolt #858: electrs connection does not work (electum and tor)

stefan-wuest, 2021-12-28:
### Describe the issue
After setting up the node based on the manual, I can not connect to my electrs server - neither with electrum nor with TOR.
I can rule out that it doesn't work because of unfinished indexing as in https://github.com/raspibolt/raspibolt/issues/723:
[image]
This block is the most recent one in realtime and eletrs indexes them one by one as they get created.

My torrc-File looks like this right now:
[torrc.txt](https://github.com/raspibolt/raspibolt/files/7785165/torrc.txt)

### Location in guide
https://raspibolt.org/electrs.html#connect-electrum-wallet](https://raspibolt.org/electrs.html#local-connection

### Screenshots
[image]

### Environment
 - Hardware platform: Raspberry Pi 4, Samsung Portable T5 2TB
 - Operating system: Raspberry Pi OS 64bit, Debian 11
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
None

VajraOfIndra, 2021-12-28:
Did you modify the Electrum config file or the shortcut target link? Could you paste the content of your Electrum Desktop config file or target link? They should require the use of one server only and contain the server IP address and port and SSL.

One thing that helped me in the past when I had troubles connecting to my Electrs was to delete a pre-existing cert:
* Shut down Electrum if it's running
* Go to `C:\Users\<yourUsername>\AppData/Roaming\Electrum\certs` (or `~/.electrum/certs` on Linux afaik)
* Delete the certificate that corresponds to your node IP address
* Check that the shorcut target link (or config file) contains the required flags/options:
  *  e.g., for the target link on Windows: `"C:\Profgram Files (x86)\Electrum\electrum-4.1.5.exe" --oneserver --server <yourNodeIPaddress>:50002:s `
* Restart Electrum

stefan-wuest, 2021-12-29:
I modified the Electrum config file.  It looks as follows:
[config.txt](https://github.com/raspibolt/raspibolt/files/7786813/config.txt)

After reading your comment I deleted the certificate under the windows folder structure mentioned and restarted Electrum. The connection is still not working.

VajraOfIndra, 2021-12-29:
The Electrum config file looks good.. I assume your node firwall has port 50002 opened. The only other thing I can think of for now is to try with the actual IP address rather than `raspibolt.local`. raspibolt.local never worked for me for anything on my network, I have to use the IP...

stefan-wuest, 2021-12-29:
The node firewall was already opened as you assumed.
You're right - I had to use the actual ip address instead.
The local connection works now. Thanks very much. I'll suggest to add this info to the guide in https://github.com/raspibolt/raspibolt/issues/747

closed 2021-12-29 by stefan-wuest

VajraOfIndra, 2022-11-18:
> > One thing that helped me in the past when I had troubles connecting to my Electrs was to delete a pre-existing cert:
> > ```
> > * Shut down Electrum if it's running
> > * Go to `C:\Users\<yourUsername>\AppData/Roaming\Electrum\certs` (or `~/.electrum/certs` on Linux afaik)
> > * Delete the certificate that corresponds to your node IP address
> > * Check that the shorcut target link (or config file) contains the required flags/options:
> >   * e.g., for the target link on Windows: `"C:\Profgram Files (x86)\Electrum\electrum-4.1.5.exe" --oneserver --server <yourNodeIPaddress>:50002:s `
> > * Restart Electrum
> > ```
> 
> Thank you, @VajraOfIndra!
> 
> Electrum Wallet guide suggestion: A troubleshooting note for those who already have Electrum installed on their systems. I needed to clear the certs directory AND use the actual local IP address to get a previously-installed Electrum to connect (for me, .local works for just about everything except Electrum).

Good idea! Could you raise an Issue with this suggestion?