raspibolt/raspibolt #859: [BUG] npm ERR when installing RTL

stefan-wuest, 2021-12-29:
### Describe the bug
After cloning the Ride The Lightning (RTL) git and checking it out, I started the installation with the command "npm install --only=prod" according to the manual.
After installing for quite a while, the below mentioned npm ERR arises.

### Location in guide
https://raspibolt.org/rtl.html#ride-the-lightning-1

### Expected behavior
Finishing installation without ERR

### Screenshots
[image]

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit, Debian 11
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
None

Stadicus, 2021-12-30:
I'm not an expert on Node.js. Maybe updating NPM helps? This is also what RTL devs usually recommend. You can try the following methods described here:

https://www.geeksforgeeks.org/how-to-update-npm/

stefan-wuest, 2022-01-01:
Thanks very much. It worked as follows (using the Admin user):
sudo npm install npm@latest -g

then proceed with the manual https://raspibolt.org/rtl.html#ride-the-lightning-1

Will add it to https://github.com/raspibolt/raspibolt/issues/747 as well in order to update the manual.

closed 2022-01-01 by stefan-wuest