raspibolt/raspibolt #860: Update wifi configuartion

sbenn9210, 2021-12-29:
### Describe the issue
A clear and concise description of where you're stuck in the guide. What are you trying to do, and how is it not working?

I am trying to update my wifi configuration after setting up the pi on one network and needing to switch it to another network. I am at the step of setting up the Electrum server in the Bitcoin section so most of the network and remote access setup has been completed on the old wifi network. What's the best way to update my wifi network?

Is the best thing to connect it to a display and keyboard to log in directly and update the wpa_supplicant.conf?

Thanks again for the help and for making the guide! I have learned a lot in the process of setting up my node. 🙏🏽 

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>

https://raspibolt.org/electrs.html

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: Raspberry Pi 4 8GB RAM
 - Operating system: 2021-10-30-raspios-bullseye-arm64-lite.img
 

### Additional context
Add any other context about the problem here.

Stadicus, 2021-12-30:
On "Raspberry Pi OS", you can run `sudo raspi-config` and select `1 System Options` --> `S1 Wireless LAN` to configure the Wifi. 

If you're wifi is currently not working, you can either by log in with SSH with an Ethernet cable, or connect the Pi to a screen/keyboard and log in locally.

[image]

VajraOfIndra, 2022-02-22:
Can you give an update on the status of this issue?

closed 2022-04-30 by github-actions[bot]