raspibolt/raspibolt #863: [BUG] Wrong link to bitcoin core used in guide (AARCH64 instead of ARM) ?

viertelb, 2021-12-30:
### Describe the bug
The guide says to download `ARM 64` bit from this site: https://bitcoincore.org/en/download/
The site provides this raw link: https://bitcoincore.org/bin/bitcoin-core-22.0/bitcoin-22.0-aarch64-linux-gnu.tar.gz
The guide also points to this site when talking about updating bitcoin core: https://bitcoincore.org/bin/bitcoin-core-22.0/
The guide then uses this ARCH64 image.

### Location in guide
https://raspibolt.org/bitcoin-core.html#installation
https://raspibolt.org/bitcoin-core.html#for-the-future-upgrade-bitcoin-core

### Expected behavior
It should download the tarball for Raspi4, which is ARM. 

However: I tried to download the version the guide later on refers to (AARCH64). **After installing bitcoind it fails to run.**

```
admin@RPi4:/tmp $ tar -xvf bitcoin-22.0-aarch64-linux-gnu.tar.gz
bitcoin-22.0/
bitcoin-22.0/README.md
bitcoin-22.0/bin/
bitcoin-22.0/bin/bitcoin-cli
bitcoin-22.0/bin/bitcoin-qt
bitcoin-22.0/bin/bitcoin-tx
bitcoin-22.0/bin/bitcoin-util
bitcoin-22.0/bin/bitcoin-wallet
bitcoin-22.0/bin/bitcoind
bitcoin-22.0/bin/test_bitcoin
bitcoin-22.0/include/
bitcoin-22.0/include/bitcoinconsensus.h
bitcoin-22.0/lib/
bitcoin-22.0/lib/libbitcoinconsensus.so
bitcoin-22.0/lib/libbitcoinconsensus.so.0
bitcoin-22.0/lib/libbitcoinconsensus.so.0.0.0
bitcoin-22.0/share/
bitcoin-22.0/share/man/
bitcoin-22.0/share/man/man1/
bitcoin-22.0/share/man/man1/bitcoin-cli.1
bitcoin-22.0/share/man/man1/bitcoin-qt.1
bitcoin-22.0/share/man/man1/bitcoin-tx.1
bitcoin-22.0/share/man/man1/bitcoin-util.1
bitcoin-22.0/share/man/man1/bitcoin-wallet.1
bitcoin-22.0/share/man/man1/bitcoind.1
admin@RPi4:/tmp $ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-22.0/bin/*
[sudo] password for admin: 
admin@RPi4:/tmp $ bitcoind --version
-bash: /usr/local/bin/bitcoind: **cannot execute binary file: Exec format error**

```
### Environment
- Raspberr Pi 4
- Raspbian GNU/Linux 10 (buster)
- Linux RPi4 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

### Additional context
I am not 100% sure since I cannot think that this issue would go unrecognised. So maybe I have the wrong OS installed? But I set up my Raspi4 according to the guide in 2020 and updated in 2021! Anyway, if there has been a change, it should be noted in the guide.

viertelb, 2021-12-30:
I seem not to be up-to-date. So maybe it is entirely my problem: 

```
admin@RPi4:~/rust/electrs $ sudo apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]                                                                                           
Hit:3 http://linux-packages.resilio.com/resilio-sync/deb resilio-sync InRelease                                                                                  
Get:4 https://deb.nodesource.com/node_12.x buster InRelease [4,584 B]                                                                                            
Get:5 https://deb.torproject.org/torproject.org buster InRelease [3,527 B]                                                 
Get:6 https://apt.syncthing.net syncthing InRelease [15.1 kB]                                        
Get:7 https://apt.syncthing.net syncthing/stable armhf Packages [1,021 B]
Get:8 https://deb.nodesource.com/node_12.x buster/main armhf Packages [780 B]
Reading package lists... Done   
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
E: Repository 'https://deb.torproject.org/torproject.org buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```

viertelb, 2021-12-30:
Turns out I am running a 32bit OS on 64bit hardware. I wonder why.

I added `arm_64bit=1` to `/boot/config.txt` and restarted and now I run 64bit:

```
admin@RPi4:~ $ uname -a
Linux RPi4 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux
```

Credits: https://www.bastianoso.de/tipps-tricks/raspberry-pi/raspberry-pi-os-auf-64-bit-umstellen.html

viertelb, 2021-12-30:
Actually, this trick just enables a 64bit kernel. The system is still 32bit and will not execute the 64bit bitcoin binary. But it will not tell you. It will instead just say: `-bash: /usr/local/bin/bitcoind: No such file or directory`

Did an earlier version of the RaspiBolt guide used a 32bit operating system?

VajraOfIndra, 2021-12-31:
> Did an earlier version of the RaspiBolt guide used a 32bit operating system?

Yes, RaspiBolt v1 and v2 were using RasPiOS Lite 32-bit. v3 has a couple of major changes: 1) it is using RasPiOS Lite 64-bit; 2) it installs the OS on the SSD and does not use the microSD card anymore

If you already had a RaspiBolt v2 running before and want to migrate to v3, you'll need to install a brand new 64-bit OS and re-install everything using the arm64 binaries. [Here are a few pointers](https://raspibolt.org/faq.html#can-i-update-my-raspibolt-2-to-the-new-version) to do the migration (how to copy the blochain and LND data to the new drive/OS)

closed 2021-12-31 by viertelb

viertelb, 2021-12-31:
I am upgrading `npm` with `sudo npm i -g npm` from version 12 to 14 to be on the same level as the guide. Leaving this here as a reminder and for comments if it is wrong to do this after we installed just with `sudo apt install nodejs´.