raspibolt/raspibolt #864: [BUG] "Warning: some journal files were not opened due to insufficient permissions." and "cannot open debug.log for reading"

orenfromberg, 2021-12-31:
### Describe the bug
Following the guide here [at this point](https://raspibolt.org/bitcoin-core.html#verification-of-bitcoind-operations), I ran the command `systemctl status bitcoind.service` and saw the following results and warning:

```
admin@raspibolt3:~/.bitcoin $ systemctl status bitcoind.service
* bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2021-12-30 23:15:37 EST; 5min ago
    Process: 786 ExecStart=/usr/local/bin/bitcoind -daemon -pid=/run/bitcoind/bitcoind.pid -conf=/home/bitcoin/.bitcoin/bitcoin.conf -datadir=/home/bitcoin/.bitcoin -startupnotify=chmod g>
   Main PID: 787 (bitcoind)
      Tasks: 17 (limit: 4165)
        CPU: 6.324s
     CGroup: /system.slice/bitcoind.service
             `-787 /usr/local/bin/bitcoind -daemon -pid=/run/bitcoind/bitcoind.pid -conf=/home/bitcoin/.bitcoin/bitcoin.conf -datadir=/home/bitcoin/.bitcoin -startupnotify=chmod g+r /home>

Warning: some journal files were not opened due to insufficient permissions.
```

Additionally, I was unable to tail the log file:
```
admin@raspibolt3:~/.bitcoin $ tail -f /home/bitcoin/.bitcoin/debug.log
tail: cannot open '/home/bitcoin/.bitcoin/debug.log' for reading: Permission denied
tail: no files remaining
```

### Location in guide
https://raspibolt.org/bitcoin-core.html#verification-of-bitcoind-operations

### Expected behavior
I expect to not see the warning or the error when tailing debug.log

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
Add any other context about the problem here.

c-mannix, 2021-12-31:
See here: https://github.com/raspibolt/raspibolt/issues/853

But I think the guide should either be updated to show the warning like mentioned in the thread linked above or the permissions updated so there are no warnings. 

For the log you need to either use 'sudo' or change the permission of the log file. Should probably update the guide for this as well because it's leads to confusion.

orenfromberg, 2021-12-31:
> See here: #853
>
> But I think the guide should either be updated to show the warning like mentioned in the thread linked above or the permissions updated so there are no warnings.
>
> For the log you [...]

I agree about showing the warning (since I was confused) and I also agree about updating the guide to use `sudo`. I will close my PR #865 and make a new one. Thank you!

PR #866 (2022-02-04): Update bitcoin-core.md: include warning and sudo when tailing debug.log
#### What

Adding an expected warning as well as prepending a command to use sudo.

This PR fixes #864 and succeeds #865 based on [this comment](https://github.com/raspibolt/raspibolt/issues/864#issuecomment-1003272244) by @c-mannix 

### Why

This change will prevent future confusion by users inspecting the bitcoind service and tailing the debug.log file.

#### How

* Now including the warning in the output
* Explicitly directing use of `sudo` with tail command
* 
#### Scope

- [ ] significant change to core configuration [... cut]
closed 2022-02-04 by VajraOfIndra