raspibolt/raspibolt #867: [lncli] unable to generate seed: rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused

xpsx, 2021-12-31:
hello everybody, I receive the following error when I try to create the first wallet:

`lnd@raspibolt:~ $ lncli create
Input wallet password: 
Confirm password: 

Do you have an existing cipher seed mnemonic or extended master root key you want to use?
Enter 'y' to use an existing cipher seed mnemonic, 'x' to use an extended master root key 
or 'n' to create a new seed (Enter y/x/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase): 
Confirm password: 

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"`

VajraOfIndra, 2021-12-31:
This error message generally means that LND is not running (and therefore the client `lncli` can't connect to the server)

Did you run "lnd" in another SSH session? If so, could you paste your LND logs to see what could prevent LND to start?

xpsx, 2021-12-31:
`lnd@raspibolt:~ $ lnd
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2021-12-31 11:22:29.531 [INF] LTND: Version: 0.14.1-beta commit=v0.14.1-beta, build=production, logging=default, debuglevel=info
2021-12-31 11:22:29.532 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2021-12-31 11:22:29.536 [INF] RPCS: RPC server listening on 127.0.0.1:10009
2021-12-31 11:22:29.541 [INF] RPCS: gRPC proxy started at 127.0.0.1:8080
2021-12-31 11:22:29.542 [INF] LTND: Opening the main database, this might take a few minutes...
2021-12-31 11:22:29.542 [INF] LTND: Opening bbolt database, sync_freelist=true, auto_compact=true
2021-12-31 11:22:29.570 [INF] CHDB: Not compacting database file at /home/lnd/.lnd/data/graph/mainnet/channel.db, it was last compacted at 2021-12-31 08:28:39.440959738 +0100 CET (2h53m50s ago), min age is set to 168h0m0s
2021-12-31 11:22:29.584 [INF] CHDB: Not compacting database file at /home/lnd/.lnd/data/chain/bitcoin/mainnet/macaroons.db, it was last compacted at 2021-12-31 08:28:39.456201637 +0100 CET (2h53m50s ago), min age is set to 168h0m0s
2021-12-31 11:22:29.591 [INF] CHDB: Not compacting database file at /home/lnd/.lnd/data/graph/mainnet/sphinxreplay.db, it was last compacted at 2021-12-31 08:28:39.467819924 +0100 CET (2h53m50s ago), min age is set to 168h0m0s
2021-12-31 11:22:29.599 [INF] CHDB: Not compacting database file at /home/lnd/.lnd/data/graph/mainnet/wtclient.db, it was last compacted at 2021-12-31 08:28:39.478777295 +0100 CET (2h53m50s ago), min age is set to 168h0m0s
2021-12-31 11:22:29.604 [INF] LTND: Creating local graph and channel state DB instances
2021-12-31 11:22:29.647 [INF] CHDB: Checking for schema update: latest_version=24, db_version=24
2021-12-31 11:22:29.655 [INF] LTND: Database(s) now open (time_to_open=111.981207ms)!
2021-12-31 11:22:29.655 [INF] LTND: We're not running within systemd
2021-12-31 11:22:29.655 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

`

VajraOfIndra, 2021-12-31:
The logs look good!

LND takes a few seconds to open.. could you try this:
* With user "lnd", run `lnd` in a SSH window
* Wait to see this line in the logging output: `Use lncli create to create a wallet, lncli unlock to unlock an existing wallet, or lncli changepassword to change the password of an existing wallet and unlock it.` (i.e. the last line of your log above)
* Then open another SSH window, log in with the `lnd` user and run `lncli create`

If it still does not work, could you paste the logs again with the error messages that should appear.

xpsx, 2021-12-31:
works perfectly, thanks!

closed 2021-12-31 by xpsx