raspibolt/raspibolt #875: [BUG] Electrs Tor hidden service redirects to port 50002 instead of 50001 (?)

VajraOfIndra, 2022-01-09:
### Describe the bug
In the guide, the Tor hidden service for Electrs contains `HiddenServicePort 50002 127.0.0.1:50002` which redirects any requests to local host on port 50002 (which is the Nginx proxy server for Electrs).

For the Explorer and RTL, the Tor hidden services do not redirect to the Nginx proxy servers (ports 4000 and 40001) but to the original servers instead (ports 3002 and 3000).

Shouldn't the Electrs Tor Hidden Service also redirect to the Electrs server directly on port 50001? so `HiddenServicePort 50002 127.0.0.1:50001` (and therefore we should also probably change it to `HiddenServicePort 50001 127.0.0.1:50001` just for consistency)

Not sure how it all works yet, but just wanted to raise the question!

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/electrs.html#remote-access-over-tor-optional)

### Expected behavior
Electrs Tor hidden service to redirect to port 50001 of local host

### Additional context
Check discussion starting here in the Telegram group: [https://t.me/raspibolt/5859](https://t.me/raspibolt/5859)

Stadicus, 2022-01-11:
This is a deliberate choice, to use the Electrs with SSL on port `50002`, even over Tor. It's not a necessity, you can also just proxy it to port `50001` without SSL.

marcosdub, 2022-01-13:
> This is a deliberate choice, to use the Electrs with SSL on port `50002`, even over Tor. It's not a necessity, you can also just proxy it to port `50001` without SSL.

Then, why do you put in the rest Tor configurations normal port? without SSL?? 

`https://raspibolt.org/btcrpcexplorer.html#remote-access-over-tor-optional`

Thanks!

Stadicus, 2022-01-17:
> Would run this torrc config instead?:

Yes, that should run as well.

closed 2022-01-18 by Stadicus

VajraOfIndra, 2022-01-21:
> This is a deliberate choice, to use the Electrs with SSL on port `50002`, even over Tor. It's not a necessity, you can also just proxy it to port `50001` without SSL.

Ok. But then why don't we do the same for the Explorer and RTL? (i.e. direct the Tor hidden service to the Nginx proxies for the Explorer and RTL). I'm sure there is a good reason.. but I can't see it yet.. could you explain why it's not necessary for RTL and the Explorer but is recommended for Electrs?

Stadicus, 2022-01-21:
Tor already contains a certificate & encryption, but some wallets require SSL. And there's no disadvantage in always using the SSL port.

For browser-based web stuff, using SSL with a self-signed certificate has a clear disadvantage in that you get scary browser warnings. So in these case, I think just using Tor is good enough and SSL does not really add much additional value.

VajraOfIndra, 2022-01-21:
> 

Very interesting, thanks for taking the time to explain.