raspibolt/raspibolt #878: Error: Specified data directory "/home/bitcoin/.bitcoin" does not exist.

jb-flow, 2022-01-09:
### Describe the bug
RaspiBolt v3 - I cannot solve the problem that the created symbolic link is not seen. The system is on SD and the HD contains the folder data with swapfile. The btcrpcexplorer service also needs to be restarted.

This is the second raspberry that I create, the first machine did not give me problems, I checked carefully for comparison I cannot understand how to solve.
```

admin@raspibolt:~ $ sudo journalctl -f -u bitcoind
-- Journal begins at Sat 2021-10-30 14:43:13 CEST. --
Jan 09 16:22:19 raspibolt systemd[1]: Starting Bitcoin daemon...
Jan 09 16:22:20 raspibolt bitcoind[559]: Error: Specified data directory "/home/bitcoin/.bitcoin" does not exist.
Jan 09 16:22:20 raspibolt systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Jan 09 16:22:20 raspibolt systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jan 09 16:22:20 raspibolt systemd[1]: Failed to start Bitcoin daemon.
Jan 09 16:23:14 raspibolt systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 1.
Jan 09 16:23:14 raspibolt systemd[1]: Stopped Bitcoin daemon.
Jan 09 16:23:14 raspibolt systemd[1]: Starting Bitcoin daemon...
Jan 09 16:23:14 raspibolt bitcoind[742]: Bitcoin Core starting
Jan 09 16:23:14 raspibolt systemd[1]: Started Bitcoin daemon.
```

```
admin@raspibolt:~ $ cd /home/bitcoin/
admin@raspibolt:/home/bitcoin $ ls -la
total 36
drwxr-xr-x 3 bitcoin bitcoin 4096 Jan  9 15:02 .
drwxr-xr-x 5 root    root    4096 Jan  9 10:58 ..
-rw------- 1 bitcoin bitcoin 1098 Jan  9 15:05 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Jan  9 10:58 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Jan  9 10:58 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   13 Jan  9 15:01 .bitcoin -> /data/bitcoin
drwxr-xr-x 3 bitcoin bitcoin 4096 Jan  9 14:16 .local
-rw-r--r-- 1 bitcoin bitcoin 1670 Jan  9 10:58 .mkshrc
-rw-r--r-- 1 bitcoin bitcoin  807 Jan  9 10:58 .profile
```

### Location in guide
https://raspibolt.org/bonus/raspberry-pi/boot-from-microsd-card.html

### Expected behavior
With due caution and restarting the services everything works. Something is lost in the startup sequence creating the alert brought to your attention.

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit

openoms, 2022-01-11:
basically you were missing a `/` when creating the symlink and now 
`.bitcoin` is linked as 
`/data/bitcoin` instead of 
`/home/bitcoin/.bitcoin`

# remove the symlink:
sudo rm /data/bitcoin

# create the directory 
sudo -u bitcoin mkdir  /data/bitcoin

# create symlink
sudo ln -s /home/bitcoin/.bitcoin   /data/bitcoin/

# check symlink:
sudo ls -la /data/bitcoin/.bitcoin

# contents
sudo ls -la /data/bitcoin/.bitcoin/
```

jb-flow, 2022-01-11:
Thanks, unfortunately it is not the solution I have done everything and reviewed the symbolic links everything works regularly as follows 

```
admin@raspibolt:/home/bitcoin $ sudo ls -la /data/bitcoin/.bitcoin
lrwxrwxrwx 1 bitcoin bitcoin 22 Jan 11 17:19 /data/bitcoin/.bitcoin -> /home/bitcoin/.bitcoin
```
As anticipated I have two raspberry, the other works correctly if I launch the command it replies 

```
admin@raspibolt:/home/bitcoin $ sudo ls -la /data/bitcoin/.bitcoin
ls: cannot access '/data/bitcoin/.bitcoin': No such file or directory
```

here is the only problem that btcexplorer fails to align so I have to restart it ...
at this point I don't think it represents a bug but something of my machine since for the other everything goes perfectly.

Stadicus, 2022-01-11:
> basically you were missing a / when creating the symlink and now
> `.bitcoin` is linked as
> `/data/bitcoin` instead of
> `/home/bitcoin/.bitcoin`

The bitcoin data directory is `/data/bitcoin`, there should be no `.bitcoin` below that. I'm not sure why bitcoind does not recognize the symlink.

From the top of my head:
* check if the external drive `/data` is mounted at `/data` with `sudo df -h`
* recreate `/data/bitcoin` and try writing to it with `touch /data/bitcoin/testfile`
* check that the link points to a valid destination (check with user "bitcoin" if it's shown in red)
  [image]

* make sure that bitcoind is run with user "bitcoin"
* make sure that `/data/bitcoin` is fully owned by user "bitcoin" with `sudo chown -R bitcoin:bitcoin /data/bitcoin`

jb-flow, 2022-01-12:
Thanks, I observe that there is something that although everything is "apparently" perfect between permissions, users, HD mounted correctly, symbolic links ... they are seen correctly but, probably they are interpreted badly, creating a malfunction. I am satisfied that a raspi with a step-by-step guide did not cause me any problems, I try to analyze better, I update you.

jb-flow, 2022-01-16:
Now I have clearer ideas, it is not an installation problem but, it derives from the inability to read the information from the HD mounted following a blackout of the machine without UPS. Having two machines but with the difference of the USB communication interface, the first has the ability to restore the status quo, while the other loses information even if correctly exposed in the file system.

VajraOfIndra, 2022-02-22:
Did you sort out your issue?

closed 2022-04-30 by github-actions[bot]