raspibolt/raspibolt #880: [BUG] Raspberry Pi 4 - RaspiOS ARM64 - Bitcoin Core aarch64 vs. arm

seaKyou1, 2022-01-15:
Hi, I am new here. Not sure if this can be classified as a bug but I tried to setup my Raspberry PI 4 following your guide and came across one small issue.

I followed the guide step by step. Downloaded and installed 2021-10-30-raspios-bullseye-arm64-lite.zip on my Raspberry and then I downloaded bitcoin-22.0-aarch64-linux-gnu.tar.gz.

After running
$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-22.0/bin/*
$ bitcoind --version

I was prompted by an error: "cannot execute binary file: Exec format error"

I realized that my kernel-version on my Raspberry Pi was armv7l by default instead of aarch64.

Following these steps
- sudo nano /boot/config.txt
- add: arm_64bit=1
- sudo shutdown -r now
I changed my kernel-version to aarch64 but I still could not run the downloaded bitcoind

So, what I did instead was downloading https://bitcoincore.org/bin/bitcoin-core-22.0/bitcoin-22.0-arm-linux-gnueabihf.tar.gz which work just fine.

Not sure if I am the only one who faced this issue and if the guide could/should be updated.

Any thoughts on why bitcoin core 32 bit on my Raspberry Pi 4 is not ideal or 64 bit would be better?

The guide has been awesome otherwise. Much appreciated, thank you!

Stadicus, 2022-01-18:
Are you setting up a new node? Then you must make sure to use the 64-bit version of the Raspberry Pi Operating System. This is what the current version of the guide uses. All binaries must be used in the 64bit version. There's no reason to use an outdated 32bit operating system if you're setting up a new node, as this will lead to certain issues, especially with LND that creates files that are too big for an old OS to handle.

If you already have a 32bit node running, you can continue to do so, but you need to download the ARMv7 32bit binaries. But sooner or later, this will cause issues with LND.

VajraOfIndra, 2022-02-22:
Did you sort out your 32 vs 64-bit OS issue?
If so, are you ok to close this issue?

closed 2022-04-30 by github-actions[bot]

seaKyou1, 2022-05-07:
> Are you setting up a new node? Then you must make sure to use the 64-bit version of the Raspberry Pi Operating System. This is what the current version of the guide uses. All binaries must be used in the 64bit version. There's no reason to use an outdated 32bit operating system if you're setting up a new node, as this will lead to certain issues, especially with LND that creates files that are too big for an old OS to handle.
> 
> [image]
> 
> If you already have a 32bit node running, you can continue to do so, but you need to download the ARMv7 32bit binaries. But sooner or later, this will cause issues with LND.
> 
> I'm not sure what the changes to " /boot/config.txt" do exactly, but you can't simply switch from 32 to 64bit like this (unless this is an awesome hack not yet known to me :). You need to install the proper operating system from scratch.

> Are you setting up a new node? Then you must make sure to use the 64-bit version of the Raspberry Pi Operating System. This is what the current version of the guide uses. All binaries must be used in the 64bit version. There's no reason to use an outdated 32bit operating system if you're setting up a new node, as this will lead to certain issues, especially with LND that creates files that are too big for an old OS to handle.
> 
> [image]
> 
> If you already have a 32bit node running, you can continue to do so, but you need to download the ARMv7 32bit binaries. But sooner or later, this will cause issues with LND.
> 
> I'm not sure what the changes to " /boot/config.txt" do exactly, but you can't simply switch from 32 to 64bit like this (unless this is an awesome hack not yet known to me :). You need to install the proper operating system from scratch.

Thank you for your reply! I am going to set up a new node and follow your advice. Hopefully that helps :)