raspibolt/raspibolt #881: Stuck at Raspberry Pi/Security/4 Increase your open files limit....

uskrusader, 2022-01-18:
### Describe the issue
A clear and concise description of where you're stuck in the guide. What are you trying to do, and how is it not working?

I'm stuck at Raspberry Pi/Security/4 Increase your open files limit....

$ sudo nano /etc/pam.d/common-session 
I can edit but not save, it says:
File 'etc/pam.d/common-session' is unwritable 

and when I try to save:
[ Error writing /etc.pam.d/common-session: Permission denied ]

please help

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>
Raspberry Pi/Security/4 Increase your open files limit (step 2

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4,, ... ]
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]
 - Version: [ output of `lsb_release -d` ]

### Additional context
Add any other context about the problem here.

Stadicus, 2022-01-18:
Looking closely at the error message, it says "Error writing /etc.pam.d/common-session" instead of "Error writing /etc/pam.d/common-session". 

Are you sure there isn't a typo and you really used the right command?

```sh
sudo nano /etc/pam.d/common-session
```

uskrusader, 2022-01-18:
no typo im afraid

![Screenshot (2)](https://user-images.githubusercontent.com/77032141/1499
[image]
66478-747f1750-c112-4ec4-9bc4-d12711d474b9.png)
d

uskrusader, 2022-01-18:
I was able to edit and save the previous instruction:

admin@raspibolt:~ $ sudo nano /etc/security/limits.conf 

with no problems, but the next two steps in "common-session" and "common-session-interactive" are "unwritable"

??

uskrusader, 2022-01-19:
How critical are these steps? Can I bypass this for now, continue, and maybe fix it later?

Stadicus, 2022-01-19:
> How critical are these steps? Can I bypass this for now, continue, and maybe fix it later?

These two adjustments are not critical. My only worry is that this could be a symptom of an underlying issue. But if everything else works, I'd just go ahead.

uskrusader, 2022-01-19:
Thanks for the input, it's all brand new equipment, never used before, I'm wondering if there's anyway I could have made a mistake in the previous instructions.  I'm a total noob, inspired by your Tales from the Crypt episode.

uskrusader, 2022-01-20:
UPDATE:  I was able to update and save the "open files limit"/common-session steps by going directly to the raspberry PI with monitor and keyboard.  

I was unable to write/save remotely using putty.

closed 2022-01-20 by uskrusader