raspibolt/raspibolt #884: .cookie missing in bitcoin data directory

ericsupreme, 2022-01-20:
### Describe the issue
i completed the generate access credentials and i don't have .cookie file in the data dir. that in turn is causing me issues with electrums running 
### Location in guide
https://raspibolt.org/bitcoin-core.html

### Screenshots
n/a

### Environment
 - Hardware platform: raspberry pi 400
 - Operating system: raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

any help will be greatly appreciated let me know if you need more info.

VajraOfIndra, 2022-01-21:
Let's check first if the .cookie file has been created and what kind of permissions it has. Could you paste the output of the following command (with user "admin"):
`$ ls -la /data/bitcoin`

ericsupreme, 2022-01-21:
yeah sorry i forgot to mention that the rpcuth.py did not create one (if i'm understanding of the docs)

[image]

Stadicus, 2022-01-21:
Do you have "rpcuser" / "rpcpassword" configured in your "bitcoin.conf"?

You can check with `cat /data/bitcoin/bitcoin.conf`

ericsupreme, 2022-01-21:
i do for the admin bc i couldn't run bitcoin-cli logged in as admin after running rpcauth.py

VajraOfIndra, 2022-01-21:
> yeah sorry i forgot to mention that the rpcuth.py did not create one (if i'm understanding the docs correctly)

Yes, the `.cookie` file is missing.

When you ran the following command `python3 rpcauth.py raspibolt YourPasswordB`, what error message did you get? Did you replace `YourPasswordB` by your own strong password?

---

By the way, and another unrelated note, you have a `bitcoin` symlink in this directory, owned by "root".. this shouldn't be there afaik.. There should only be three `.bitcoin -> /data/bitcoin` symlinks:

1. In the "admin" user root folder `/home/admin`, with the symlink being owned by "admin"
2. In the "bitcoin" user root folder: `/home/bitcoin`, with the symlink being owned by "bitcoin"
3. In the "lnd" user root folder: `/home/lnd`, with the symlink being owned by "lnd"

ericsupreme, 2022-01-21:
i didn't receive any error message that i know of and i replaced with my own password. it returned the output and a copied it to bitcoin.conf per the docs. 

and yes buddy i definitely screwed up the symlinks as well lol i tried going back to rm the link and fixing but don't think i did it right i may just start the pi build from scratch. do you think that is necessary or can i salvage this? just curious, thanks.

Stadicus, 2022-01-21:
> i do bc the admin couldn't run bitcoin-cli logged in as admin after running rpcauth.py

If you use `rpcuser`/`rpcpassword`, Bitcoin Core will not create the .cookie file. These are mutually exclusive.

If user "admin" has access to the `.cookie` file, e.g. through a symlink `/home/admin/.bitcoin --> /data/bitcoin` , `bitcoin-cli` should detect that.

ericsupreme, 2022-01-21:
okay and thank you guys i really appreciate it! i'm going to just go thru the docs step by step again to make sure all is config correctly.

ericsupreme, 2022-01-21:
> If you use `rpcuser`/`rpcpassword`, Bitcoin Core will not create the .cookie file. These are mutually exclusive.
>
> If user "admin" has access to the `.cookie` file, e.g. through a symlink [...]
yeah see when i comment out

> If you use `rpcuser`/`rpcpassword`, Bitcoin Core will not create the .cookie file. These are mutually exclusive.
>
> If user "admin" has access to the `.cookie` file, e.g. through a symlink [...]

yeah see when i comment out rpcuser/rpcpasssword i get "error: Could not locate RPC credentials. No authentication cookie could be found, and RPC password is not set.  See -rpcpassword and -stdinrpcpass.  Configuration file: (/home/admin/.bitcoin/bitcoin.conf)"

is that do to my screwed up symlinks?

ericsupreme, 2022-01-21:
hey i think commenting out rpcuser/rpcpasssword did the trick b/c i can now run bitcoin-cli as other users and now i just need to undo whatever i did to electrs config because i'm getting a different error message (see below) now which is promising.
[image]

Stadicus, 2022-01-21:
Seems like Electrs is already running, and cannot reserve the same port for itself again. Make sure that it is only started on boot (only one systemd unit file) and restart the Pi. Maybe that's already enough.

ericsupreme, 2022-02-07:
this issue can be closed thanks you.  electrums was indeed running and started indexing.  the issue was resolved when i stopped trying to use rpc / rpcpassword and rpcauth as described above.

closed 2022-02-07 by ericsupreme