raspibolt/raspibolt #885: [FEATURE REQ] Update lntop bonus guide to v0.3.1

VajraOfIndra, 2022-01-21:
### What & why

Just a note as a reminder for a contributor to update the lntop bonus guide to update it from v0.2.0 to v0.3.1: [https://github.com/edouardparis/lntop/releases](https://github.com/edouardparis/lntop/releases)

marcosdub, 2022-01-26:
There aren't binaries avaliable for v0.3.1 arm64 arch yet. Better v0.3.0, avaliable binaries:
https://github.com/edouardparis/lntop/releases/tag/v0.3.0
https://github.com/edouardparis/lntop/releases/download/v0.3.0/lntop-v0.3.0-Linux-arm64.tar.gz

VajraOfIndra, 2022-01-26:
> There aren't binaries avaliable for v0.3.1 arm64 arch yet. Better v0.3.0, avaliable binaries: https://github.com/edouardparis/lntop/releases/tag/v0.3.0 https://github.com/edouardparis/lntop/releases/download/v0.3.0/lntop-v0.3.0-Linux-arm64.tar.gz

Good point, updating to v0.3.0 instead sounds good.

PR #934 (2022-02-23): Update lntop to v0.3.0 + new Update & Uninstall sections
#### What

* Updates the lntop bonus guide to latest released binary (from v0.2.0 to v0.3.0)
* Updates heading level to match rest of the guide
* Adds an "Update" and "Uninstall" sections

### Why

To keep the lntop guide up-to-date

#### Scope

- [ ] significant change to core configuration
- [x] independent bonus guide
- [ ] simple bug fix

Fixes #885 

#### Test & maintenance

Install lntop from scratch to check that the updated guidelines are ok.
Update and uninstall lntop to check the thw 2 new sections are ok

[image]
closed 2022-02-23 by VajraOfIndra