raspibolt/raspibolt #887: Electrs not indexing, stuck in fail/restart cycle.

uskrusader, 2022-01-24:
### Describe the issue
Electrs not indexing, it is stuck in a fail/restart cycle, repeating over and over gain.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/electrs.html#autostart-on-boot)
$ sudo journalctl -f -u electrs   
(its been running for hours, but now constantly failing/restarting.

### Screenshots
[image]

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit, Debian 11

### Additional context
Add any other context about the problem here.

I'm not sure what to do next?!

VajraOfIndra, 2022-01-24:
It looks like your "electrs" user does not have the permission to read the Bitcoin Core `.cookie` file.

Could you paste the output of the following command (with user "admin"):
`$ sudo ls -la /data/bitcoin`

And also the output of:
`$ groups electrs`

(The first command will tell if your `.cookie` file exists and is readable by members of the group "bitcoin", while the second command will tell if the "electrs" user is part of the "bitcoin" group.)

uskrusader, 2022-01-24:
[image]

uh-oh!  I don't see a cookie file, did I miss a step? ...or do a step in the wrong folder/directory? 
Thanks for helping me with this

uskrusader, 2022-01-24:
[image]

I redid previous steps, not sure if this helps...

VajraOfIndra, 2022-01-25:
> [image]
> 
> uh-oh! I don't see a cookie file, did I miss a step? ...or do a step in the wrong folder/directory? Thanks for helping me with this

Ok so your "electrs" user is part of the "bitcoin" group, so that's great.

But you are missing the `.cookie` file indeed. Also, you have a symlink in `/data/bitcoin ` which should not be there. @ericsupreme in #884 has the same issue.. so I'm starting to wonder if there is a wording issue somewhere in the guide that leads to this. @Stadicus, any thoughts on this recurring issue?

In your `bitcoin.conf` file (open it with `sudo nano /home/bitcoin/.bitcoin/bitcoin.conf`), in the "# Connections" section, do you have the following line (but with the `raspibolt:01234....abcd` replaced by your own string generated by the Python script):
`rpcauth=raspibolt:01234....abcd`

Did you use the guide's template ([here](https://raspibolt.org/bitcoin-core.html#configuration)) to create your `bitcoin.conf` file?

uskrusader, 2022-01-26:
I followed the guide, but I did have a couple of connection interruptions....

[image]

uskrusader, 2022-01-26:
...just redid and noticed this:

[image]

VajraOfIndra, 2022-01-26:
> I followed the guide, but I did have a couple of connection interruptions.... Does this look right?
> 
> (aside from the 2 added (#) to reduce 'dbcache' after full sync))

Yes, the conf file looks good.

VajraOfIndra, 2022-01-26:
> ...just redid this step and noticed this:

That's because you're trying to change the permissions with your "admin" user who does not have the rights to do this file.

Either use the same command with your "bitcoin" user, as is done in the guide (or else with user "admin" try using adding `sudo` in front of the command, it should work as well).

Once you've done that, start bitcoind again (or restart it if it's already running as a background service) and check if the cookie file has been created.

uskrusader, 2022-01-26:
feeling dumb here.... so what exactly is the command to restart (or reindex)

[image]

uskrusader, 2022-01-26:
it looks like it running in the background, yes?

[image]

VajraOfIndra, 2022-01-26:
The errors in the logs could be due to launching bitcoind while it's already running in the background... but that's just an uninformed guess...

If `bitcoind` is running in the background (as it seems to be), that's what I would do:
* Restart bitcoind by issuing the following command (with the "admin" user): `sudo systemctl restart bitcoind`
* Wait a few minutes
* Check the logs by issuying the following command: `sudo tail -f /data/bitcoin/debug.log`
  * do the logs look ok? do they contain error messages?
* If the logs look ok, check that the `.cookie` file has been created by checking the content of the Bitcoin folder: `sudo ls -la /data/bitcoin`

uskrusader, 2022-01-27:
Looks like errors....  it seems to continuous retry until I CTRL-C

[image]

VajraOfIndra, 2022-01-27:
> Looks like errors.... it seems to continuous retry until I CTRL-C
> 
> "Please restart with -reindex or -reindex-chainstate to recover" (I'm not sure what the command line input would be for that if it is required)

I never encountered this issue before so I hope someone else might help you better than me.. but this is what I would try (there might be a better/quicker way, I'm just telling you how I'd do it with the little I know :). Basically I'd stop and disbale the service and start Bitcoin Core manually since I know I can add a flag to the command, so:
```
$ sudo systemctl stop bitcoind
$ sudo systemctl disable bitcoind
$ sudo su - bitcoin
$ bitcoind -reindex-chainstate
```
Wait for the reindexing to be completed and see if bitcoind then runs fine:
* if yes, proceed to check if the `.cookie` file exists as described above
* if not, stop bitcoind (Ctrl+C) and start it again but this time with the -reindex flag (it will take longer)
  *  `bitcoind -reindex`

Once you're finished (if all works well), make sure you stop bitcoin (Ctrl+C) and then enable and start the service again:
```
$ sudo systemctl enable bitcoind
$ sudo systemctl start bitcoind
```

I hope someone else might be able to shed some light on your issues though.

uskrusader, 2022-02-04:
[image]

Am I good to go here?  Is my Electrs fully indexed?

VajraOfIndra, 2022-02-05:
> Am I good to go here? Is my Electrs fully indexed?

Looks good to me! I have the same kind of outputs in my logs and my Electr is fully indexed.

VajraOfIndra, 2022-02-22:
Can we close this issue? :)

uskrusader, 2022-03-01:
close it!  ....

closed 2022-03-01 by VajraOfIndra