raspibolt/raspibolt #888: [MOD REQ] Remove line "index_looup_limit" of electrs.conf

marcosdub, 2022-01-24:
### What & why

Remove parameter "index_looup_limit" of electrs.conf, now limit is 200 txs by default.

In the last released of electrs: https://github.com/romanz/electrs/blob/master/RELEASE-NOTES.md#094-dec-30-2021 ... "Disable index_lookup_limit by default (635)" now, the number of transactions to lookup before returning an error, has risen to 200 txs in source code. (evindences below). Is not necessary.

### How

Remove parameter "index_lookup_limit" of electrs.conf:
```
# RaspiBolt: electrs configuration
# /data/electrs/electrs.conf

# Bitcoin Core settings
network = "bitcoin"
daemon_dir= "/home/bitcoin/.bitcoin"
daemon_rpc_addr = "127.0.0.1:8332"
daemon_p2p_addr = "127.0.0.1:8333"

# Electrs settings
electrum_rpc_addr = "127.0.0.1:50001"
db_dir = "/data/electrs/db"

# Logging
log_filters = "INFO"
timestamp = true`
```
### Additional context

**Discussions:**

https://github.com/romanz/electrs/issues/635
https://github.com/romanz/electrs/pull/637

**Evidences & example devs conf:**

https://github.com/romanz/electrs/blob/cd348e72de9071549755170e9709ae922e023083/internal/config_specification.toml#L111-L115

https://github.com/romanz/electrs/blob/master/doc/config_example.toml

Stadicus, 2022-01-25:
Sounds like a good change :+1:

marcosdub, 2022-01-26:
Opening error

PR #1049 (2022-07-11): Remove outdated "index_lookup_limit" from config
#### What/why/how

The Electrs configuration file option `index_lookup_limit` is now disabled by default because not useful for operators using Electrs for themselves only. 

See #888 for more detailed explanations and useful links.

#### Scope

- [ ] significant change to core configuration
- [x] minor change to core configuration
- [ ] independent bonus guide
- [ ] simple bug fix

Fixes #888 

#### Test & maintenance

Run Electrs with the deleted conf option

#### Animated GIF (optional)
[image]
closed 2022-07-11 by VajraOfIndra