raspibolt/raspibolt #892: Mismatched Ports in Config

xmready, 2022-01-25:
There is a mismatch between the port numbers in the bitcoin and electrs config files. I don't know if this is on purpose or a typo in the guide.

Bitcoin config shows ports 28332 & 28333 
`zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333`

electrs config sows ports 8332 & 8333
`daemon_rpc_addr = "127.0.0.1:8332"
daemon_p2p_addr = "127.0.0.1:8333"`

I'm guessing that rpc is for local connections and p2p is for other nodes? Thanks.

Stadicus, 2022-02-01:
These are not the same services provide by these ports:

* `8332`: Bitcoin Core RPC API (to accept queries, e.g. from `bitcoin-cli`)
* `8333`: Bitcoin Core P2P network (to connect to other Bitcoin nodes)
* `28332`/`28333`: [ZeroMQ](https://zeromq.org/) ports used by LND to get notifications on new blocks and stuff

xmready, 2022-02-01:
> These are not the same services provide by these ports:
>
>     * `8332`: Bitcoin Core RPC API (to accept queries, e.g. from `bitcoin-cli`)
>
>     * `8333`: Bitcoin Core P2P network (to connect to other [...]

Thanks for the clarification. Maybe the guide could use a top level explanation, or some commented lines in the config.

closed 2022-02-11 by xmready