raspibolt/raspibolt #894: Prefer Fulcrum to Electrs

craigraw, 2022-02-01:
In #646, I suggested replacing Electrs with ElectrumX, largely due to bugs present in Electrs at the time. Even though the performance of ElectrumX is demonstrably faster on deeper wallets, this suggestion was (correctly) rejected due to the poor indexing performance of ElectrumX, particularly on the 32 bit OS used by Raspibolt at the time. Indexing performance on a 64 bit OS is better, but still not enough to recommend it on an RPi4.

I believe Fulcrum is a more scalable and performant choice than Electrs (or ElectrumX), as discussed here: https://www.sparrowwallet.com/docs/server-performance.html. Although the index is bigger, it will fit easily on the the 1TB disk recommended by Raspibolt. The guide already uses `txindex=1` on Bitcoin Core, so there is no additional space used there. The only downside then is slightly longer indexing time (currently 2-3 days instead of 1, depending on the `fast-sync` setting), but I think the advantages are significant. Much faster queries, more reliable data (Electrs falls over with deep wallets and serves bad data), and likely prolonged hardware lifespan.

In addition, [binaries for arm64](https://github.com/cculianu/Fulcrum/releases/tag/v1.6.0) are available - no more compiling with Rust! I would love to see Fulcrum become the default Electrum server for my favourite node setup :)

Stadicus, 2022-02-01:
Thanks for this input and your ongoing research in Electrum server performance. A very interesting read. 

I'm open to switching to Fulcrum eventually if it's really that much faster. I'm not sure if a hard replacement is a priority just now. Maybe we can start with a bonus guide (mentioning it as an alternative in the Electrs section) and decide which server to go forward long-term on the next major release?

RooSoft, 2022-02-01:
Thanks for sharing that invaluable info @craigraw! You got me curious about Fulcrum.

craigraw, 2022-02-02:
My primary reason for suggesting this change is more about data integrity than performance. I get at least one query every day (usually more) on the Sparrow Telegram group relating to Electrs server issues. Some of these may be Tor-related, but I believe the majority are simply due to Electrs itself. As discussed in the report, the data architecture of Electrs is such that it puts significant strain on an SBC for all but the most simple of wallets. This leads to Electrs failing to notify of transactions affecting a wallet during normal operation, and eventually entering a moribund state where it does not respond to requests at all, requiring a restart. To overcome these issues I believe Electrs will need to adopt an entirely different data architecture, which I think is unlikely in the short to medium term.

RooSoft, 2022-02-06:
I've had an extensive chat with Calin Culianu... very productive, this guy really wanted to help.

It can be condensed to this:

- They'd be very excited to have Fulcrum added to our guides
- Fulcrum has been created and is maintained with Bitcoin Cash in mind
- They say keeping compatibility with Bitcoin Core is trivial and very important to them

Nothing more to add, really... aside from the fact that this is the first line of their README.md

`A fast & nimble SPV server for Bitcoin Cash & Bitcoin BTC.`

That's an opinion, but having mixed feeling about such cringe. Also, if for some reason Bitcoin Cash changes enough to make Fulcrum not compatible with Bitcoin anymore, we could have a problem on our hands.

VajraOfIndra, 2022-05-22:
A bonus guide is being prepared: #1021

closed 2022-07-28 by github-actions[bot]