raspibolt/raspibolt #896: [FEATURE REQ] Split "Electrum Server" guide into two guides: "Electrum Server" & "Desktop wallet" + focus on one desktop app only (e.g. Sparrow)

VajraOfIndra, 2022-02-02:
### What & why

Presently, the Electrum server guide contains instructions on:
1. How to install and setup Electrs
2. How to connect the Electrum wallet and BitBoxApp to Electrs

However, the Electrums server is used for other apps than Electrum and BitBoxApp: e.g. the BTC RPC Explorer, Bisq and possibly many other apps.
I'm noticing that new users often have some issues connecting their desktop wallet (e.g. Electrum) to their Electrum server. This could be partly due to the concise nature of the existing guidelines.

Therefore, to make things clearer and to have a bit more space to explain how to connect a wallet to the sevcer, I think that separating the two sections could make a lot of sense, i.e.:
1. An "Electrum server" guide that focuses solely on the installation and configuration of Electrs (and pointing to alternative in the bonus section, e.g. ElectrumX etc)
2. Followed by a "Desktop wallet" guide that clearly explains how to install and connect a desktop wallet to Electrs

So the new layout for the "Bitcoin" section of the guide would be:
1. Bitcoin Core*
2. Electrum server
3. Desktop wallet
4. Blockchain Explorer

* _Note_: as the other sub-headings are all generic names rather than program names.. we should maybe use "Bitcoin node implementation" or something more generic like this rather than "Bitcoin Core".. wdyt?

For the "Desktop wallet" guide, I think we should focus on explaining how to install and connect only one popular desktop wallet. All other apps could be as bonus guides linked to in this section. IMHO, the two main candidates would be either:
1. Electrum
2. Sparrow

My preference would go to Sparrow as its focus on showing how things are under the hood really resonates with the whole "RaspiBolt" philosophy (+ its native coinjoining capabilities that could be highlighted)

### How

- Restrict the existing "Electrum Server" guide to installation and configuration only (remove the Electrum and BitBoxApp sections)
- Create a new "Desktop wallet" guide that explains how to install only one popular app (e.g. Sparrow) and how to connect it to Electrs. Links to the bonus section for guides on how to connect other apps (Electrum, BitBoxApp, Specter, Bisq etc).

Wdyat?

VajraOfIndra, 2022-02-02:
This change would place emphasis on installing a bitcoin desktop app (so on another device than the SCB), which could be seen (or not) as straying away from a guide explaining how to setup a SCB-based node.

It really depends what goals we want the guide to achieve:
1. Do we want to keep it solely focused on having a bitcoin and lightning node on a SCB? in that case a bitcoin desktop wallet is seen as secondary and would belong to bonus guides or optional sections... or
2. Do we want the guide to provide an integrated set of self-sovereign tools to bitcoiners? in that case, a guide that explains how to setup a desktop wallet and connect it to the node would prefectly fit in

IMHO, I like to think of the RaspiBolt as more option 2. But I'd understand if vision number 1 is preferred and therefore a separate bitcoin desktop wallet guide would'nt really make sense.

In any case, even if vision 1 is preferred, I believe the Electrum Server guide would benefit from focusing on only one wallet, have that section a bit expanded to clarify a few common mistakes and then having links to bonus guides for other apps.

Anyways, enough of me talking. Looking forward to hear feedback and ideas on all this :)

VajraOfIndra, 2022-02-07:
PS: forgot to say: I'd be happy to have a go at writing a draft PR for this... but I wanted to see if there was some appetite for it before, as it would take a fair amount of time. Hence the thread here.

VajraOfIndra, 2022-02-13:
Just to give an idea on how it could look, here is a first draft: https://vajraofindra.github.io/RaspiBolt/

zynos, 2022-02-14:
I like the separation and also that you have chosen sparrow wallet as an examlpe :)

uskrusader, 2022-03-24:
Is there any issue connecting BOTH electrum and sparrow wallets to my electrum server using the exact same xyzaddress.onion:50002 -and- proxy: localhost:9050 ?

Or is this the only possible way to connect them?.... can both wallets run at the same time using these ports/proxys?  I would think so, but I'm still very new to all this stuff.... thanks!

VajraOfIndra, 2022-04-07:
> Is there any issue connecting BOTH electrum and sparrow wallets to my electrum server using the exact same xyzaddress.onion:50002 -and- proxy: localhost:9050 ?

Afaik, it should be ok.

> Or is this the only possible way to connect them?.... 

You could connect Sparrow to Bitcoin Core directly also. But it 's more complicated as it requires to edit the bitcoin.conf file. It also leave a wallet.dat unecnrypted on your node which means anyone that get read access to your node (ok, unlikely..) could know your public keys and balance. See https://www.sparrowwallet.com/docs/best-practices.html#the-expert-stage

> can both wallets run at the same time using these ports/proxys? I would think so, but I'm still very new to all this stuff.... thanks!

The proxy for Sparrow is not related to your node. It determines wheter Sparrow will use clearnet or Tor to get two pieces of information: 1) the bitcoin price from a public API, 2) the communication with the Whirlpool server when doing coinjoins

closed 2022-04-29 by VajraOfIndra