raspibolt/raspibolt #907: [FEATURE REQ]

zynos, 2022-02-09:
### What & why

> As a tutorial reader, I want a link to the bitcoin user creation so that i dont need to panic when i should `Allow the user “bitcoin” to configure Tor directly` before i created the user.

### How
Here you give instructions on how to change the "bitcoin" users rights:
https://raspibolt.org/privacy.html#configuration

But in the guide, the user is created in the next chapter:
https://raspibolt.org/bitcoin-core.html#create-the-bitcoin-user

I think it would be nice to link to that chapter already.

Thank you for the guide so far!

RooSoft, 2022-02-09:
Great catch... that user is now being created farther in the guides, in the Bitcoin Section.

This needs to be addressed ASAP

Stadicus, 2022-02-10:
Thanks for the feedback, we just recently moved the creation of user "bitcoin" into the "Bitcoin Core" section. This issue is an unwanted consequence.

The command `sudo adduser bitcoin debian-tor` simply needs to be moved to the "Bitcoin Core" section as well.

closed 2022-02-10 by Stadicus