raspibolt/raspibolt #911: Initial Block Download Isssue

penguin4111, 2022-02-10:
### Describe the issue
I have gone through the latest RaspiBolt guide and got to the point where you wait on the initial block download. That was a little over a month ago. I am currently on progress=0.860373 for blocks during January of 2021. It has slowed down significantly and I am getting a new block every 8 minutes roughly. At this rate I may never catch up to the live chain. 

I am running a 1TB HDD through USB3 connection. When I did the speed test that the guide instructed me to do, I got a speed of around 80MB/s which was above the stated 50 that the guide said should be fast enough. Additionally, I have my Pi hooked up through a direct ethernet line rather than wifi. 

One clue I see is that when I run "bitcoin-cli getnetworkinfo" it only ever shows 1 outgoing connection and sometimes shows 0. Any idea why or what things I might check? I was under the impression this number should be closer to 10 connections. Maybe this is why things are so slow?

### Location in guide
(https://raspibolt.org/bitcoin-core.html)

### Environment
 - Hardware platform: Raspberry Pi 4 4GB
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

RooSoft, 2022-02-10:
Seems like a connectivity issue to me... have you tried to reboot the node?

Maybe also try some I/O with it by downloading a file from the internet...

Also, is it using wired ethernet or wifi?

penguin4111, 2022-02-14:
Okay so I guess you were right, it was a connectivity issue. I had to unplug and replug the ethernet cable from the Pi and I guess that was all it needed. No idea why because the cable was definitely not loose or anything. But now it works MUCH better so anyways its fine.

closed 2022-02-14 by penguin4111