raspibolt/raspibolt #914: debug.log shows many failed connections

xmready, 2022-02-11:
Recurring failed connection lines in Bitcoin debug.log happen every few minutes, different IPs. Using tor proxy 9050, configs are the same as the guide. Blockcount appears to be current.

https://raspibolt.org/bitcoin-core.html#verification-of-bitcoind-operations

Lines from debug.log
`Socks5() connect to xxx.xxx.xxx.xxx:8333 failed: general failure`
`Socks5() connect to  v3TorAddress:8333 failed: host unreachable`

Output of bitcoin-cli getnetworkinfo
>{
  "version": 220000,
  "subversion": "/Satoshi:22.0.0/",
  "protocolversion": 70016,
  "localservices": "0000000000000409",
  "localservicesnames": [
    "NETWORK",
    "WITNESS",
    "NETWORK_LIMITED"
  ],
  "localrelay": true,
  "timeoffset": -4,
  "networkactive": true,
  "connections": 15,
  "connections_in": 5,
  "connections_out": 10,
  "networks": [
    {
      "name": "ipv4",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    },
    {
      "name": "ipv6",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    },
    {
      "name": "onion",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    },
    {
      "name": "i2p",
      "limited": true,
      "reachable": false,
      "proxy": "",
      "proxy_randomize_credentials": false
    }
  ],
  "relayfee": 0.00001000,
  "incrementalfee": 0.00001000,
  "localaddresses": [
    {
      "address": "v3TorAddress.onion",
      "port": 8333,
      "score": 4
    }
  ],
  "warnings": ""
}

### Environment
 - Hardware platform: AMD Ryzen 3900x, 4GB RAM
 - Operating system: Ubuntu Server 64bit LXC Container in Proxmox
 - Version: Ubuntu 20.04.3 LTS

RooSoft, 2022-02-12:
Can I see your bitcoin.conf?

RooSoft, 2022-02-16:
Can you send the output of `sudo lsof -i -P -n | grep LISTEN`, especially lines starting by `bitcoind`

RooSoft, 2022-02-16:
Also...

```bash
$ sudo su - bitcoin
```

To switch to the `bitcoin` user and

```bash
$ groups
```

To list `bitcoin` user's groups...

`debian-tor` must be part of the output, if not, run this with the `admin` user

```bash
$ exit
$ sudo adduser admin bitcoin
```

And restart bitcoin core...

```bash
sudo systemctl restart bitcoind.service
```

xmready, 2022-02-16:
> Seems to me that your node is listening on port 8333 and the bitcoin user can use tor...
> 
> Not a tor expert, maybe someone else could help about that...

The failed connections are happening with both Tor v3 and IPv4. I consistently have 10 outgoing connections, my block count is current, and I can remotely connect to electrs via Tor v3. I'm not so sure the problem is with Tor at all..

RooSoft, 2022-02-16:
Maybe someone could go to the LND slack and ask them... might be part of normal operation... 🤷

xmready, 2022-02-16:
> Maybe someone could go to the LND slack and ask them... might be part of normal operation... shrug

I haven't installed LND on this node. It's just bitcoind and electrs for now. Do you see these same lines in your debug.conf?

RooSoft, 2022-02-16:
Yeah, I got some... sorry, might ask Bitcoin Core then...

xmready, 2022-02-16:
Yea, looks like Bitcoin Core is my next stop. Thanks

RooSoft, 2022-02-16:
Would be very useful if you could write your findings here... thanks!

closed 2022-04-24 by github-actions[bot]