raspibolt/raspibolt #915: [BUG]

zynos, 2022-02-13:
### Describe the bug
Link to next section is not working.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/static-channel-backup.html#option-1-local-backup)

### Expected behavior
The link:
https://raspibolt.org/static-channel-backup.html#(optional)-remote-backup
should bring me to the next section.

### Proposed solution
replace the old link with an updated one like:
https://raspibolt.org/static-channel-backup.html#option-2-remote-backup-preparations

NaincyKumariKnoldus, 2022-02-13:
Hey! @zynos I have tried to update these links. Please review my PR.
Thanks!

VajraOfIndra, 2022-02-17:
The link is now fixed. Are you happy to close this Issue?

zynos, 2022-02-17:
The new link does not work for me:
see https://github.com/raspibolt/raspibolt/commit/4cb430e37a1092345efff4ff158873c563a2f163#r66890346

VajraOfIndra, 2022-02-17:
> The new link does not work for me: see [4cb430e#r66890346](https://github.com/raspibolt/raspibolt/commit/4cb430e37a1092345efff4ff158873c563a2f163#r66890346)

Good point, was it in the previous Issue and PR? If so, I completely missed it! 
I'll write a PR now to fix it.

closed 2022-02-17 by zynos

VajraOfIndra, 2022-02-17:
Fixed, the PR has been merged :)