raspibolt/raspibolt #916: [BUG] Guide Bonus Section: Connect to Zeus outdated

zynos, 2022-02-13:
### Describe the bug
The .lnd link is not located under /home/bitcoin .
It is located under /home/lnd

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/bonus/lightning/zeus-over-tor.html)

### My solution
Replace
`sudo su bitcoin` with `sudo su lnd`
and Replace
`lndconnect --lnddir=/home/bitcoin/.lnd --host=z1234567890abc.onion --port=8080`
with
`lndconnect --lnddir=/home/lnd/.lnd --host=z1234567890abc.onion --port=8080`

I was able to connect my Zeus App under **IOS** (testflight environment) sucessfully afterwards!
I dont know if Android works as well,

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit

VajraOfIndra, 2022-02-17:
Thanks for reporting these bugs. The Zeus bonus guide is indeed outdated and not compatible with v3.

However, we are in the process of making Zeus part of the core guide, see pull-requst #921. This PR should be merged in the coming weeks hopefully. Once merged, the bonus guide will not exist anymore.

Rather than correcing the bonus guide (which display the "not tested on v3" label as a warning), I propose we simply wait for #921 to be merged. 

Wdyat?

uskrusader, 2022-03-10:
I'm stuck in the Zeus guide. The QR code is massive and I can't shrink it. Ctrl + - is not working in the putty termininal. It works for most of my other windows, but not putty where that ginormous QR code is. I Cant fit it on screen to scan.  8-(

Stadicus, 2022-03-10:
@uskrusader, I created a new issue for this: https://github.com/raspibolt/raspibolt/issues/959

Stadicus, 2022-03-10:
This issue can be closed, as the bonus guide has been replaced with the "Mobile App" section in the core guide:
https://raspibolt.org/guide/lightning/mobile-app.html

closed 2022-03-10 by Stadicus