raspibolt/raspibolt #919: [BUG] Bonus guide: Mempool - Nginx fails to start on reboot.

ghgeiger, 2022-02-14:
### Describe the bug
After successful install and start of mempool.space on RPI4/2G - SD/SSD raspibolt 3, upon reboot, Nginx fails to start and prevents all we interfaces from operating.

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/bonus/bitcoin/mempool.html)

### Expected behavior
Nginx and all web services function after reboot.

### Screenshots
If applicable, add screenshots to help explain your problem.
[image]

### Environment
 - Hardware: Raspberry Pi 4 - 2GB, OS on 64 GB microSD, Data on 1TB SSD
 - Operating system: Raspberry Pi OS 64 bit
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
It seems that this occurred because nginx was expecting mempool to be operational, but mempool had not started yet because it is waiting for bitcoind to be functional. 

Rectified by changing the file:
`/lib/systemd/system/nginx.service`
to include the lines:
```
Restart=on-failure
RestartSec=30s 

```
under the line:
`Type=forking`

RooSoft, 2022-04-09:
Can you paste the output of that command here please?

```bash
cat -n /etc/nginx/snippets/nginx-mempool.conf | sed -n 45,55p
```

RooSoft, 2022-04-09:
Mine contains that on line 49 and above... is yours similar?

```ini
    49		# mainnet API
    50		location /api/v1/donations {
    51			proxy_pass https://mempool.space;
    52		}
```

RooSoft, 2022-04-09:
`Host not found` can also be a DNS problem, as this URL seems to redirect to an external site... maybe also try this

```bash
ping mempool.space
```

and make sure it returns a valid IP

wiz, 2022-05-06:
@RooSoft this is my fault, we need to add a `resolver 8.8.8.8;` line in nginx.conf to enable dynamic on-demand DNS resolution, by default it tries to resolve names on startup which fails for most people because their network / internet connectivity is not working yet and it causes nginx to fail starting

RooSoft, 2022-05-06:
Is there a way to prevent this?

ghgeiger, 2022-05-06:
Sorry, I seem to have missed a bunch of notifications on this thread. Unfortunately, I uninstalled memspace in an effort to reduce load on my node, so I can't provide any further feedback.

closed 2022-07-12 by github-actions[bot]

wiz, 2022-07-13:
FWIW we fixed this by deleting those lines from the nginx.conf so this is fixed in mempool v2.4.0

erikarvstedt, 2022-07-13:
The fix is in https://github.com/mempool/mempool/commit/fdb035c0d244de15e512906ea8c3e8227311cf9e.

VajraOfIndra, 2022-07-14:
I will make a PR to update Mempool to version 2.4.0.