raspibolt/raspibolt #922: Error with lnd going from RaspiBolt 2 to 3

penguin4111, 2022-02-16:
### Describe the issue
I was previously running a lightning node with channels using the RaspiBolt 2 guide. I am now reconfiguring my node using the RaspiBolt 3 guide and am running into an issue. I got to the point where I installed and initially ran lnd - that was all fine. I then took my thumbdrive which had my old node's data directory and mounted it and used the command in the FAQ section to transfer it over. That seemed to work fine (from what I could tell). The guide doesn't really say what to do from there but I went ahead and started up lnd at that point and I get the following error:

```
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2022-02-16 12:58:02.858 [INF] LTND: Version: 0.14.2-beta commit=v0.14.2-beta, build=production, logging=default, debuglevel=info
2022-02-16 12:58:02.859 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2022-02-16 12:58:02.860 [ERR] LTND: Shutting down because error in main method: unable to load TLS credentials: tls: private key does not match public key
2022-02-16 12:58:02.860 [INF] LTND: Shutdown complete

unable to load TLS credentials: tls: private key does not match public key
```

Please let me know if there is something I missed or am potentially doing wrong. Thanks!

### Location in guide
(https://raspibolt.org/faq.html)

### Environment
 - Hardware platform: Pi 4, 4 gb
 - Operating system: Raspberry Pi OS 64bit
 - Version: Debian GNU/Linux 11 (bullseye)

VajraOfIndra, 2022-02-17:
Could you check who is the owner of the `/data/lnd` directory? (e.g. with `ls -la /data`)

You might need to change the ownership to user "lnd" with:
`sudo chown -R lnd:lnd /data/lnd`

penguin4111, 2022-02-17:
Okay so I ran `la -la` and I believe the output suggests that there is a symbolic link connecting `.lnd` and `/data/lnd` (see below). But when I actually go into the respective folders `~/.lnd` and `~/data/lnd` I find that the folder contents are NOT the same. I am really not sure why?

```
lnd@raspibolt:~ $ ls -la
total 36
drwxr-xr-x 4 lnd  lnd  4096 Feb 16 12:54 .
drwxr-xr-x 6 root root 4096 Feb 16 12:29 ..
-rw------- 1 lnd  lnd   644 Feb 17 11:36 .bash_history
-rw-r--r-- 1 lnd  lnd   220 Feb 16 12:29 .bash_logout
-rw-r--r-- 1 lnd  lnd  3523 Feb 16 12:29 .bashrc
lrwxrwxrwx 1 lnd  lnd    13 Feb 16 12:30 .bitcoin -> /data/bitcoin
drwxr-xr-x 3 root root 4096 Feb 16 12:57 data
lrwxrwxrwx 1 lnd  lnd     9 Feb 16 12:30 .lnd -> /data/lnd
drwxr-xr-x 3 lnd  lnd  4096 Feb 16 12:31 .local
-rw-r--r-- 1 lnd  lnd  1670 Feb 16 12:29 .mkshrc
-rw-r--r-- 1 lnd  lnd   807 Feb 16 12:29 .profile
```

VajraOfIndra, 2022-02-17:
> Okay so I ran `ls -la` and I believe the output suggests that there is a symbolic link connecting `.lnd` and `/data/lnd` (see below). 

So here we are in your "lnd" user home directory.
The `lrwxrwxrwx 1 lnd  lnd    13 Feb 16 12:30 .bitcoin -> /data/bitcoin` and `lrwxrwxrwx 1 lnd  lnd     9 Feb 16 12:30 .lnd -> /data/lnd` are the symlinks to the Bitcoin Core and LND data directories, owned by "lnd". So that's all good.

However, there is a `data` folder there which, afaik, should not be here.. which brings us to your next question.

> But when I actually go into the respective folders `~/.lnd` and `~/data/lnd` I find that the folder contents are NOT the same. I am really not sure why?

* When moving to `~/.lnd` with your "lnd" user you go to the following absolute path: `/home/lnd/.lnd` which translates to `/data/lnd`
* When moving to `~/data/lnd` with your "lnd" user you go to the absolute path: `/home/lnd/data/lnd`. This is probably a folder you pasted/sent to the wrong location.

> To answer your question as well, when I run `la -la /data/lnd` I get the following output

Regarding the content of `/data/lnd`, all the files and directories are owned by the "lnd" user so that's a good start.

However, this symlink `lrwxrwxrwx 1 lnd   lnd      9 Feb 17 11:36 lnd -> /data/lnd` is wrong. It should be a real folder containing all the LND databases and data, not a symlink as it is now.

> The contents I believe are a mish-mosh of files from my old and new node. lit.conf, for example, was an old .conf file I was using in my old node setup.

Yes, looks like there was a couple of mis-steps along the way when you copy-pasted the old LND data on your new node...

If you still have your old data on the thumbdrive, I would probably try to delete the `/data/lnd` folder on your new node and then copy/paste your old `lnd` folder. WARNING: before deleting any lnd data folders, make sure you have a backup or copy of your old LND data somewhere! If you inadvertently delete all your copies of your old data, you won't be able to recover your node! :)

Assuming you have a safe copy of the old LND data somewhere, I would do this with user "admin":
 
First, remove/back-up the problematic folders:  
  ```sh
  $ sudo mv /data/lnd /data/lnd.bak
  $ sudo mv /home/lnd/data / /home/lnd/data.bak
  ```

Then I'd copy over the old LND data to the new node, either using the [method described in the guide](https://raspibolt.org/faq.html#backup--restore-lnd-data). Or else by doing it on a laptop (I've done it this way when migrating from v2 to v3, but it's probably longer and less efficient):
 * I would shut down LND and bitcoind and shut down the node and remove the SSD.
 * Then I'd plug the SSD and thumbdrive in a laptop and move over the old LND data to the new node with: `sudo rsync -rhvP --append-verify /path/to/thumbdrive/data/lnd /path/to/new/node/drive/data`
 * Then I'd plug the SSD drive into the Pi and plug the power to start the Pi
 * I'd then change the ownership of the lnd data folder: `sudo chown -R lnd:lnd /data/lnd`

As you can see my methid is more involved and time-consuming. Might be bette to use the method describied in the guide's FAQ!

Then try to start LND see if it works. 

If it works well, later on you could delete the folder that we backed-up as they are not needed, i.e.:
* `/data/lnd.bak`
* `/home/lnd/data.bak`

Good luck!

closed 2022-04-25 by github-actions[bot]