raspibolt/raspibolt #924: inadvertently used usb 2 port for sata drive

ericsupreme, 2022-02-16:
hello. i inadvertently used usb 2 port for my sata ssd and when i try to plug it into the usb 3.0 (blue port) the rasp pi will not boot. is there  anyway i can fix this and move drive to usb 3.0 ? thanks

ericsupreme, 2022-02-16:
> hello. i inadvertently used usb 2 port for my sata ssd and when i try to plug it into the usb 3.0 (blue port) the rasp pi will not boot. is there anyway i can fix this and move drive to usb 3.0 ? thanks

in the meantime i'm going to try this Bonus Guide: Upgrade External Drive

VajraOfIndra, 2022-02-17:
> hello. i inadvertently used usb 2 port for my sata ssd and when i try to plug it into the usb 3.0 (blue port) the rasp pi will not boot. is there anyway i can fix this and move drive to usb 3.0 ? thanks

I don't know how to help you but I posted your question to the Telegrqm group in case someone there has some advice for you.

ericsupreme, 2022-02-17:
> I don't know how to help you but I posted your question to the Telegrqm group in case someone there has some advice for you.

thanks i appreciate i!! i think the answer has something to do with etc/fstab and unmounting and remounting but i'm nervous lmao

visskiss, 2022-02-21:
I don't imagine that if you're using a boot from USB as outlined in this guide that there should be any issue at all switching ports. My guess is something is wrong with your drive or with the usb 3 slot. Can you still start up from usb 2 slot?

ericsupreme, 2022-03-02:
> I don't imagine that if you're using a boot from USB as outlined in this guide that there should be any issue at all switching ports. My guess is something is wrong with your drive or with the usb 3 slot. Can you still start up from usb 2 slot?

sorry for the long delay, it just will not boot from usb 3. when i switch back to 2 it works just fine. i am using raspberry pi 400. i'm about the try pi 4.  thanks

ericsupreme, 2022-03-02:
> quick question before i close, i am only getting 33mb per second after running sudo hdparm -t --direct /dev/sda and of course the doc says 50mb+ is there any thing i can do to increase that on the usb2 port? the link to Check the [Fix bad USB3 performance](https://raspibolt.org/troubleshooting.html#fix-bad-usb3-performance) in that partiular section is broken.

sweet!!!!! i went to waybackmachine.com  to access the broken link and usb-storage.quirks got me up to 284mb's a second!! whooaaaa!!! thanks again

closed 2022-03-02 by ericsupreme

Stadicus, 2022-03-02:
Cool, great to hear. We just published a big change to the file structure, and the old link probably didn't work anymore. It does now, though. Here's the correct one:
https://raspibolt.org/guide/troubleshooting.html#fix-bad-usb3-performance