raspibolt/raspibolt #930: [DISCUSSION]: Standardization of Bitcoin and Lightning sub-headings: generic names vs program names

VajraOfIndra, 2022-02-18:
## Introduction

As stated by @Stadicus in this [comment](https://github.com/raspibolt/raspibolt/pull/921#issuecomment-1044668250), it is preferrable to have generic names for the left menu headings:

> Regarding "Generic" vs. "Application-specific" titles: I strongly prefer the generic titles, at least when there's a choice. This also affects the filenames, and we can replace individual applications in the core guide easier if necessary.

It does indeed make the guide more future-proof. However, presently, there are still a couple of headings that derive their names from the chosen program to install on the node, namely:
* Bitcoin Core
* LND
* Ride The Lightning

NB: Interestingly I assume "Electrum server" has now become a generic name! E.g. Fulcrum could be categorized as an Electrum server, correct? I looked around to see if there was a more generic name without the term 'Electrum' in it but coulnd't find anything at all. Any thoughts on this? Feel freet to bring this in the discussion below and propose alternative names.

NB 2: In this post, I'll focus on the "Bitcoin" and "Lightning" chapters and will ignore the "Raspberry Pi" chapter of the guide, although the heading "Raspberry Pi" is more specific than generic... Feel freet to bring this in the discussion if you wish so.

---

## New names proposals

I propose below a few ideas to homogenize the left menu headings with the following goals:
* *Generic rather than specific*: All headings are have generic names and not program names
* *Industry standard*: The generic name are well-known and accepted categories in the bitcoin community
* *Short*: The names are kept short for ease of display and navigation

### Program: Bitcoin Core

* _Existing_: **Bitcoin Core**
* _Proposed_:  **Bitcoin Node**
* _Alternative_: **Bitcoin Node Implementation** -> Quite long (also not an implementation per se as there is no specifications, correct?.. but that's pedantry now :)
* Alternative: **Bitcoin Client** -> but it's more than just a client...
* Alternative: **Bitcoin Core** -> we keep it as is, see [comment](https://github.com/raspibolt/raspibolt/pull/921#issuecomment-1044668250). IMHO, I don't think it's needed to keep the program name as the generic names above looks suitable.

### Program: LND

* _Existing_:  **LND**
* _Proposed_: **LN Node** -> Reuse the same terminology as 'Bitcoin Node' = visually consistent
* _Alternative_: **LN Node Implementation** -> Technically better, but quite long

### Program: Ride The Lightning

* _Existing_: **Ride The Lightning**
* _Proposed_: **Web Node Manager** -> The RTL repo describes itself as "a full function, device agnostic, web user interface to help manage lightning node operations", while ThuderHub has "an open-source LND node manager where you can manage and monitor your node on any device or browser"
* _Alternative_: **Web Dashboard** -> A bit simpler and shorter than the above..

### Program: Zeus

* _Existing (soon! with Zeus)_: **Mobile Wallet** -> not ideal as Zeus is much more than just a LN wallet
* _Proposed_: **Mobile Node Manager** -> to reuse the terminology used for RTL
* _Alternative_: **Mobile Dashboard** -> idem but with dashboard which is simpler/shorter

I'll update the OP with any comments, pros/cons, and new name proposals made in the comments below!
I'll add new sections if you want to discuss other names not listed just above (Electrum server, Raspberry Pi or something else).

Wdyat?

RooSoft, 2022-02-18:
**Program: Bitcoin Core**

_Bitcoin Node Implementation_ pretty much is programming jargon to me. It means an app or library that follows a specification.

If I put myself in the shoes of a typical end user, I'd be simply looking for my _Bitcoin Node_, so that's my preference for a simple user interface.

There may be a caveat here... less computer illiterate people might consider the whole Raspibolt as a Bitcoin node... but I suspect that these guys might be a tad on the power user side so that might not be an issue.

RooSoft, 2022-02-18:
**Program: Electrs**

Seems to me that _electrum_ over time has become a protocol that wallets can use to aggregate data that Bitcoin Core doesn't. It used to be an app that has been replicated from scratch in different languages with different performance features. In the end, I think a wallet can call any of those implementations and expect to get similar, if not the exact same results.

An _Electrum server_ pretty much is an app that understands that protocol and answers through an API. That pretty much is what we're refering to here, so that's my preference.

RooSoft, 2022-02-18:
**Program: LND**

Here we're clearly referring to a _lightning network node_. Would be nice over time to allow the user to choose from a variety of implementations. Although that might affect apps depending on it as all of them expose wildly different APIs to the lightning apps.

I guess _lightning network node_ might be more future proof. Having mixed feeling calling it a _LN node_, though. Could cause confusion to some, I guess.

Stadicus, 2022-02-18:
I admire your spirit, and I'm all for a harmonized naming scheme. Reading the above, the discussion feels a bit academic, however. Above all, I think the titles must be **easy to understand**. Everything else, semantics included, should come second.

A few remarks on the individual sections:

### Bitcoin Core

The program we use is "Bitcoin Core", which is a "Bitcoin client". It's really exactly that: a client on the network. I wouldn't call it a node: this is a technical term and more associated with the overall solution/device.

### Electrum server

I think this title is fine as is. "Electrum" is the protocol, and the program "Electrs" is a server that serves information using the Electrum protocol. I wouldn't use Title Case, as this implies that "server" is part of a name.

### Blockchain Explorer

Technically, the program is called "BTC RPC Explorer", but in the UI it calls itself "Bitcoin Explorer". Following the "not using Title Case" and generic naming logic, it could be "Blockchain explorer".

### LND

I agree that this could be less application-specific. "LN" is hard to understand, however. How about "Lightning client", following the "Bitcoin client" example? We also don't call it "Bitcoin Network client"... :)

### Ride the Lightning

Functionally, I think the term "Web Node Manager" is too broad (the "node" being the overall solution), and "Web Dashboard" doesn't specify much (and implies "read-only" functionality). Maybe "Lightning web interface" would be adequate?

### Zeus

Same logic as with RTL: "Mobile Node Manager" or "Mobile Dashboard" doesn't click for me. Zeus is also not a client, it's just a user interface. Maybe "Lightning mobile interface"? 

Personally, I find the "Lightning web/mobile interface" a bit clunky and technical. The titles are in the "Lightning" section anyway, so maybe just "Web interface" and "Mobile interface"?

RooSoft, 2022-02-18:
Wow, those are great takes, and I approve most of them.

I wouldn't use the `interface` term though, as it sounds a bit power user, or developer to me...

App might be more appropriate for regular folks.

Stadicus, 2022-02-18:
> App might be more appropriate for regular folks.

Yes! 

* Web app
* Mobile app

VajraOfIndra, 2022-02-19:
So we would have:

> * Bitcoin
>   * Bitcoin client
>   * Electrum server
>   * Blockchain explorer
> * Lightning
>   * Lightning client
>   * Static channel backup
>   * Web app
>   * Mobile app

LGTM.. shall we make a PR?

Stadicus, 2022-02-22:
Sound good to me! 

On thing: I'm not quite sure if the "Static Channel backup" is too specific. It's correct, of course, but does it need to be so specific in the navigation?

How about naming this guide "Channel backup"?

VajraOfIndra, 2022-02-22:
> Sound good to me!
> 
> On thing: I'm not quite sure if the "Static Channel backup" is too specific. It's correct, of course, but does it need to be so specific in the navigation?
> 
> How about naming this guide "Channel backup"?

I like the idea of making this sub-heading more generic!

My only little nit is that using 'Channel backup' might reinforce the false premise that the user can backup their node and recreate it if something goes wrong. This is not fault of ours but due to LND initial poor choice of name for this file (it's a misnomer).

Technically the SCB is used as part of the LND "Data Loss Protection" protocol.. but it does not make much more sense either tbh since we don't protect the data but the funds..

Having said that, I'm all for using "Channel backup" as I can't think of a better alternative, unless you can think of something else.

VajraOfIndra, 2022-02-22:
@Stadicus: Just to double-check, I assume we should also rename the file names, not just the content, right? E.g. `rtl.md` would become `web-app.md`

RooSoft, 2022-02-23:
> @Stadicus: Just to double-check, I assume we should also rename the file names, not just the content, right? E.g. `rtl.md` would become `web-app.md`

Would make a lot of sense...

VajraOfIndra, 2022-02-23:
Based on latest comments, we'd have:

> * Bitcoin------------------------------> 30 `bitcoin.md`
>   * Bitcoin client--------------------> 10 `bitcoin-client.md`
>   * Electrum server----------------> 20 `electrum-server.md`
>   * Blockchain explorer-----------> 30 `blockchain-explorer.md`
> * Lightning--------------------------> 40 `lightning.md`
>   * Lightning client-----------------> 10 `lightning-client`
>   * Channel backup---------------> 20 `channel-backup.md`
>   * Web app-------------------------> 30 `web-app.md`
>   * Mobile app----------------------> 40 `mobile-app.md` #921 

Looks good, simple and logical.

I'll prepare a PR implementing these changes.

RooSoft, 2022-02-24:
I like the filenames!

Side note: I'd vote for folders if the document generator supports them.

VajraOfIndra, 2022-02-24:
> I like the filenames!
> 
> Side note: I'd vote for folders if the document generator supports them.

If I understood well what you mean, it does. Actually, it seems to be the recommended way to organize docs in the Just the Docs [guidelines](https://pmarsceill.github.io/just-the-docs/docs/navigation-structure/#pages-with-children). This is how the bonus section is done presently.

So, we could have (ignoring the images and resources folder and all the Jekyll and GitHub files):

+-- ..
|
|-- docs
|   |-- bitcoin
|   |   |-- `index.md`  (parent page, previously `bitcoin.md`)
|   |   |-- `bitcoin-client.md`
|   |   |-- `blockchain-explorer.md`
|   |   |-- `electrum-server.md`
|   |
|   |-- bonus
|   |   |-- `index.md`  (parent page)
|   |   |-- bitcoin
|   |   |   |-- `index.md`  (parent page)
|   |   |   +-- _`(all the Bitcoin bonus guides)`_
|   |   |-- lightning
|   |   |   |-- `index.md`  (parent page)
|   |   |   +-- _`(all the Lightning bonus guides)`_
|   |   |-- raspberry-pi
|   |   |   |-- `index.md`  (parent page)
|   |   |   +-- _`(all the Raspberry Pi bonus guides)`_
|   |
|   |-- lightning
|   |   |-- `index.md `     (parent page, previously `lightning.md`)
|   |   |-- `channel-backup.md`
|   |   |-- `desktop-app.md`
|   |   |-- `lightning-client.md`
|   |   |-- `web-app.md`
|   |
|   |-- raspberry-pi
|   |   |-- `index.md`      (parent page, previously `raspberry-pi.md`)
|   |   |-- `operating-system.md`
|   |   |-- `preparations.md`
|   |   |-- `privacy.md`
|   |   |-- `remote-access.md`
|   |   |-- `security.md`
|   |   |-- `system-configuration.md`
|   |
|   |-- `backstory.md`
|   |-- `faq.md`
|   |-- `troubleshooting.md`
|
|-- `index.md` (parent page, the website home page)
+-- ..

I really like the idea of using the recommended strcuture for docs. However, and if it's a desired change, it might be safer to do this a separate PR: one for the changing the names to generic name; and one for changing the structure of the repo. Wdyt?

Stadicus, 2022-02-24:
Sounds good to me. Of course, changing filenames ofter leads to broken links, so we should do it once and then it should stay fixed for a long time. This is also why I didn't want to introduce folders in the beginning, as the guide structure was not yet ossified enough. Now I think folders would be a good option.

I was thinking that some filenames could be more descriptive than their title, e.g., `lightning-web-app.md`. But if the file `web-app.md` is in the folder `lightning/` then that's good as well.

RooSoft, 2022-02-24:
I'd set the filenames as close as the title as possible so that contributors can find them easily even if new.

PR #938 (2022-02-24): Harmonize left menu headings with generic names
#### What

See discussion in #930 

_tl;dr_: The PR updates the left menu headings to make them generic rather than program-specific.

### Why

* Consistency/harmonization
* Makes it easier to switch a program with another one in the future if desired, without changing the guide's main structure

#### How

* Changed file names: `*.md`
* Changed yml titles
* Changed h1 titles
* Changed the title/path of the "Next" links at the bottom of the page

#### Scope

- [x] significant change to core configuration
- [ ] independent bonus guide
- [ ] simple  [... cut]
PR #940 (2022-03-02): Harmonize left menu headings with generic names (#938)
Follow-up change to https://github.com/raspibolt/raspibolt/pull/938#event-6131425079

Original change description by @VajraOfIndra 

---

#### What

See discussion in #930 

_tl;dr_: The PR updates the left menu headings to make them generic rather than program-specific.

### Why

* Consistency/harmonization
* Makes it easier to switch a program with another one in the future if desired, without changing the guide's main structure

### How

* Changed file names: `*.md`
* Changed yml titles
* Changed h1 titles
* Changed the title/path of th [... cut]
closed 2022-03-02 by Stadicus