raspibolt/raspibolt #933: [FEATURE REQ]  update your rsync command to include maintaining owner and group

visskiss, 2022-02-21:
### What & why

When following [this guide for transferring data from an on old install](https://raspibolt.org/faq.html#can-i-update-my-raspibolt-2-to-the-new-version), you use the rsync command with the flags 

-rhvP

You should update this to use the flags

-rhvPog

So that owner and group are maintained as well as permissions. 

Either that or you need to add the commands to update owner and group, ie

chown -r bitcoin:bitcoin /data/bitcoin

etc.

> As a [description of user], I want [functionality] so that [benefit].

### How

Describe the solution you're thinking of. Provide a clear and concise description of what you want to happen. Did you consider alternative options?

### Additional context

Add any other context or screenshots about the feature request here.

VajraOfIndra, 2022-02-22:
Good idea! 

`-ahvP` could work as well and preserve modification times (but it works without preserving them so it does not add much)

This would work great for copying over the Bitcoin data and avoid having to do a `chown` command.   
However, for copying over the LND data, we would still need to use `sudo chown -r lnd:lnd /data/lnd` since the LND data folder owner/group has changed from "bitcoin" (in v2) to "lnd" (in v3).

VajraOfIndra, 2022-02-22:
@visskiss See PR #935 Could you leave a review?

PR #935 (2022-02-24): Bitcoin & LND data migration: added -og flag to rsync command + LND chown command
#### What

See discussion in #933. The PR adds presently missing guidelines concerning the ownership of the Bitcoin and LND data folders when migrating from v2 to v3.

### Why

To improve the change of users successfully migrating their data to a new node

#### Scope

- [ ] significant change to core configuration
- [ ] independent bonus guide
- [x] simple bug fix

Fixes #933 

[image]
closed 2022-02-24 by VajraOfIndra