raspibolt/raspibolt #947: simple problem?

uskrusader, 2022-03-02:
### Describe the issue
A clear and concise description of where you're stuck in the guide. What are you trying to do, and how is it not working?
So I ran the guide all the way to the end of "Bitcoin/Electrum server" section.  My electrum wallet will not connect to my node.  So I went back through a bunch of steps in "Bitcoin Client" section.  When I try to see "Bitcoind" in action, I get an "permission denied" error.  It seems to work if I put "sudo" in front of "tail -f /home/bitcoin/.bitcoin/debug.log"   

Did I do something wrong?  or did the guide miss the "sudo" command?

### Location in guide
Raspibolt Guide: Bitcoin/Bitcoin Client

Link to the corresponding page and section on <https://raspibolt.org>
https://raspibolt.org/guide/bitcoin/bitcoin-client.html#verification-of-bitcoind-operations

### Environment
 - Hardware platform: Raspberry Pi 4
 Raspberry Pi OS 64bit, Debian 11, ... ]

### Additional context
Add any other context about the problem here.

Stadicus, 2022-03-03:
Hi there, the `sudo` command should not be necessary for user "admin" to have read-only access to the Bitcoin directory. this sounds like a permission problem.

Could you run the following command and see what your system says in regards to the highlighted part?

[image]

If you get an error, maybe you need to run `sudo ls -la /data/bitcoin` due to the same issue.

uskrusader, 2022-03-03:
Mine looks a little different...  
[image]

RooSoft, 2022-03-03:
Use the `bitcoin` user and you won't need to sudo...

Also, your `admin` user probably isn't in the `bitcoin` group... would be an even simpler fix.

```bash
sudo adduser admin bitcoin
```

And then you'll be able to tail the file directly with `admin`.

uskrusader, 2022-03-03:
Looks like i already added admin to bitcoin group (?)
...but still doesnt work without sudo 
[image]

VajraOfIndra, 2022-03-03:
**OOT:** @Stadicus, I see that your `debug.log` can also be read by "other"/any users... same with mine. Is that expected knowing that we grant access only to the group users with `chmod g+r /data/bitcoin/debug.log`...?

I would expect the file permissions to be:
`-rw-r-----`
and not:
`-rw-r--r--`

What am I missing there?

Stadicus, 2022-03-03:
> What am I missing there?

Probably nothing, the file does not need to be readable by anyone. My system is a mess... :)

VajraOfIndra, 2022-03-03:
> > What am I missing there?
> 
> Probably nothing, the file does not need to be readable by anyone. My system is a mess... :)

so is mine then :laughing: I also have -rw-r--r--

RooSoft, 2022-03-03:
Yeah.. that's definitely a `chmod` job... totally missed this detail.

uskrusader, 2022-03-03:
[image]

> Your `debug.log` file can only be read by the "bitcoin" user right now. You probably missed this step: [image]
> 
> Log in with user "bitcoin" and run the above command: `chmod g+r /data/bitcoin/debug.log`
> 
> Then, any members of the group "bitcoin" (e.g. the "admin" user) will be able to read the log.

So I ran the command above twice.  I wasn't sure if /home/admin mattered. (see pic above)
that seemed to work.

I see lots of failures when i run the original tail command: tail -f /home/bitcoin/.bitcoin/debug.log
Is that normal?

So far, my electrum wallet is connected to my raspibolt node through TOR, (1 connection) I have the blue button.  I can't seem to connect it without TOR. I have received to and sent from a small amount of bitcoin to it to test it, success. But I'm still having issues connecting it to the BTC RPC explorer with or without TOR.  I started this project with ZERO knowledge. I feel so close to the end, but I'm still plagued with these little issues.  Thanks for your all your help.

VajraOfIndra, 2022-03-03:
> So I ran the command above twice. I wasn't sure if /home/admin mattered. (see pic above) that seemed to work.

Being in /home/admin did not matter because you provided the full (absolute) path: `/data/bitcoin/debug.log` 
If you had provided only the log name `debug.log` for example, you would have needed to move into `/data/bitcoin` first with `cd /data/bitcoin`
 
> I see lots of failures when i run the original tail command: tail -f /home/bitcoin/.bitcoin/debug.log Is that normal?

It seems a common issue yes. See #914 for example. Nothing to worry about for now afaik.

> So far, my electrum wallet is connected to my raspibolt node through TOR, (1 connection) I have the blue button. I can't seem to connect it without TOR. I have received to and sent from a small [...]

Networking issues are very common yes. I suggest you close this issue and open new ones for each of your specific issue (one for your Electrum wallet connection issue and one for your BTC RPC Explorer local connection issue). 

Before opening these 2 new issues, if you didn't do it yet, maybe do a search in the repository for old issues with similar topics, you might find some helpful answers in older threads! But if not, just open new issues with the required information and you'll quickly get some help :)

closed 2022-03-04 by uskrusader