raspibolt/raspibolt #949: Node will not connect to blockchain explorer

uskrusader, 2022-03-04:
I ran through this part of the guide twice, verified the .env file, hopefully it is in the right place.  
but it still wont connect to BTC explorer for some reason.  
The last pic at bottom would indicate node is operating fine though, right? 

### Location in guide
https://raspibolt.org/guide/bitcoin/blockchain-explorer.html
[image]

[image]

regular connection
[image]

through TOR
[image]

The node is working fine right?  If electrum wallet works with one blue button tor connection with .onion addy I got by following this guide.  I did the small transaction with it only connected to the one nod.

[image]

### Environment
 Raspberry Pi 4, 
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]

Stadicus, 2022-03-05:
> The node is working fine right? If electrum wallet works with one blue button tor connection with .onion addy I got by following this guide. I did the small transaction with it only connected to the one nod.

That's working as expected, nothing to do here :)

Stadicus, 2022-03-05:
Your BTC RPC Explorer configuration file is also loaded correctly, looks good. But there's an error message about the port already being in use. Maybe the Explorer is already running in the background. I'd recommend a restart of the RaspiBolt and checking the log immediately after boot:

```sh
sudo journalctl -u btcrpcexplorer -f
```

You can also check the Bitcoin Core log if there are any error messages resulting from the Explorer trying to fetch data with wrong credentials:

```
tail -f /data/bitcoin/debug.log
```

Stadicus, 2022-03-05:
Another common issue is that the Explorer does not reload the `.cookie` information if it changes, for example, because Bitcoin Core is restarted, but the Explorer was already running.

See here: https://github.com/janoside/btc-rpc-explorer/issues/393

In that case, try restarting the Explorer systemd unit (the background service) with

```
sudo systemctl restart btcrpcexplorer
```

uskrusader, 2022-03-05:
> Another common issue is that the Explorer does not reload the `.cookie` information if it changes, for example, because Bitcoin Core is restarted, but the Explorer was already running.
> 
> See here: [janoside/btc-rpc-explorer#393](https://github.com/janoside/btc-rpc-explorer/issues/393)
> 
> In that case, try restarting the Explorer systemd unit (the background service) with
> 
> ```
> sudo systemctl restart btcrpcexplorer
> ```

that fixed it !!!
thank you much!

closed 2022-03-05 by uskrusader