raspibolt/raspibolt #952: onion address for electrum same to connect other wallets to raspibolt?

uskrusader, 2022-03-05:
Perhaps another simple question....
Do I use the same onion address I use for electrum wallet to connect node to other wallets? 
 ie Wasabi Wallet

Vielen Dank!

[image]

[image]

luneth, 2022-03-06:
No. Let me explain,
You connect your electrum wallet to your electrs instance (ie: abc....xyz.onion:50002 or yourlocalipaddress:50002) which indexes all the transactions in the blockchain and act as a "bridge" between bitcoin core and electrum (or bluewallet, etc...)

Wasabi works differently, thanks to [Bitcoin knots](https://bitcoinknots.org/) can act as a full node (meaning it also has to download the entire blockchain, which is useless if you already have a full node running at home) and it indexes it itself. If you want wasabi to connect to your node you should use  abc.....yxz.onion:8333 or yourlocalipaddress:8333

tldr: to connect to wallets using electrum's protocol => node's-adress:port 500002
to connect wasabi to your node =>  node's-adress:port 8333

uskrusader, 2022-03-06:
> 
> tldr: to connect to wallets using electrum's protocol => node's-adress:port 50002 to connect wasabi to your node => node's-adress:port 8333

So I use the same onion address, but I just change the port to 8333?

luneth, 2022-03-06:
I you are in the same local network as the node I would strongly advice you to connect using its local IP address (i.e 192.168.XXX.XXX:8333)

If you want to access it remotely use the onion address found with this command:
bitcoin-cli getnetworkinfo

This address isn't the same as the one serving elected, which can be found with this command (assuming you've followed the entire electrum server part of this guide) :
sudo cat /var/lib/tor/hidden_service_electrs/hostname

uskrusader, 2022-03-06:
> 
> If you want to access it remotely use the onion address found with this command: bitcoin-cli getnetworkinfo
> 
> This address isn't the same as the one serving electrs, which can be found with this command (assuming you've followed the entire electrum server part of this guide) : sudo cat /var/lib/tor/hidden_service_electrs/hostname

that was a big help!  That onion address matches the one in the BTC RPC explorer that I was wondering about.  I wasn't sure if that was my address or someone else's connected to mine.  Learning is happening, thank you!

luneth, 2022-03-06:
No worries 👍
BTW to find peers connected to your node (both inbound and outbound) use: bitcoin-cli getpeerinfo
Or just go to the peer section of btcrpcexplorer.

You'll find that you can do nearly everything btcrpcexplorer does using bitcoin-cli. And if you want to dive further into bitcoin technical aspects and out to use bitcoin-cli [here](https://github.com/BlockchainCommons/Learning-Bitcoin-from-the-Command-Line/blob/master/README.md) is a perfect guide.

Wishing you the best during your journey with bitcoin!

uskrusader, 2022-03-06:
> If you are within the same local network as the node I would strongly advice you to connect using its local IP address (i.e 192.168.XXX.XXX:8333) to avoid wasting tor network's bandwidth uselessly.

What is the best way to find this IP address? I seem to have more luck with onion addresses sofar. I know the local IP address of the raspberry pi (raspibolt 192.168.xx.xx), but that doesn't seem to work when I enter it into wasabi with port :8333.  (the onion address does work, I see the connection on the btc rpc explorer, and also in wasabi's log file) 

Is there a different local ip address for the node operating inside the raspberry pi?  I have the FING app, and it shows many generic IP addresses in my local wifi network....

luneth, 2022-03-06:
Your node only has one IP address on you local network. 

I believe your problems is firewall related.
If you've followed this guide by the letter you should have ufw set up to block all incoming connection except for ssh (port 22),  btcrpcexplorer (port 4000), ride the lighting (port 4001) and electrs (port 50002)

For wasabi to be able to connect to your node you need to open 
port 8333:
sudo ufw allow 8333 (to allow port 8333 from anywhere)
sudo ufw allow from 192.168.1.0/24 to any port 8333 comment 'Allow access to bitcoin-core from lan' (to allow port 8333 only from within your lan)

The first one should be enough as your router's own firewall doesn't forward incoming traffic on port 8333 to your node by default. 
The second one is a bit more cumbersome, but more secure depending on your situation.

Bonus:
It is overkill in your situation but instead of allowing an entire subnet to access port 8333 you could only authorize one device.
Let's say your computer ip is 192.168.1.53 the ufw rule would be:
sudo ufw allow from 192.168.1.53 to any port 8333 comment 'Allow access to bitcoin-core from my computer'

closed 2022-03-08 by uskrusader