raspibolt/raspibolt #953: Add i2p connectivity

luneth, 2022-03-06:
Before submitting a PR to add i2p to this guide, I would like to know how you feel about it.

Even though i don't believe the tor network to collapse anytime soon, I believe adding a second network, still protecting users anonymity, as second medium/failover wouldn't be anything but beneficial.

Thoughts?

luneth, 2022-03-07:
- The idea isn't to completly replace tor with i2p. 
Mostly because concerning Bitcoin Core, having i2p as the only medium to connect to other node could potentially lead to potential sybil attacks. Since as this moment not a lot of nodes at using i2p.

- You could serve web app/services on both network. Basically creating i2p adresses the same way you create hidden services with tor.

- The same way they actually do with tor, instead of having tor installed and accessing an abc....xyz.onion adress they would have to install i2p software and access an abc....xyz.b32.i2p adress

I believe this fits the decentralized spirit of bitcoin. Not being dependable of one network, but 3. tor, i2p and using both to anonymously reach nodes on clearnet.

Stadicus, 2022-03-08:
Sounds great! If this can be added without disrupting the existing configuration, that would be a perfect bonus guide for now. Maybe parts of it could make it into the main guide in the future.

kristapsk, 2022-03-10:
I2P should be used in addition to Tor and only for bitcoind for now. Having two different privacy networks should improve resistance against possible attacks against your node, also might be helpful for users in countries where state is actively fighting against Tor by banning Tor node IP addresses (but not sure can or can not and are they trying currently to do that against I2P).

I tried to do it some time ago, bitcoind even announces I2P address, but connections aren't working. Haven't yet managed to get back to it. See also https://old.reddit.com/r/raspibolt/comments/sdh113/add_i2p_connectivity/.

VajraOfIndra, 2022-03-15:
Congrats for making it work! This will be such a powerful addition to the core RaspiBolt setup.

Do you think it can go straight to the core guide or is it safer to propose it as a bonus guide for a while?

kristapsk, 2022-03-15:
Was initially tending towards doing it as a core guide from the start, but probably it's better to start with a bonus guide, collect some initial feedback from the users and then move it to the core.

VajraOfIndra, 2022-04-12:
Pinging @luneth :) do you want to tackle this update or are you ok with @kristapsk doing it? 

It'd be great to have this added in the not too distant future.

kristapsk, 2022-06-11:
Should not closed, it's on my TODO list.

VajraOfIndra, 2022-07-11:
> Should not be closed, it's on my TODO list.

FYI, configuration tips by @twofactor: https://t.me/raspibolt/11831

> https://i2pd.readthedocs.io/en/latest/user-guide/install/#debian
> +
> sudo systemctl enable i2pd
> +
> add to bitcoin.conf :
> 
> debug=i2p
> #onlynet=i2p #optional
> i2psam=127.0.0.1:7656
> i2pacceptincoming=1

twofaktor, 2022-07-11:
> > Should not be closed, it's on my TODO list.
> 
> FYI, configuration tips by @twofactor: https://t.me/raspibolt/11831
> 
> > Install apt-transport-https package

sudo apt-get install apt-transport-https 

Automatically add repository

wget -q -O - https://repo.i2pd.xyz/.help/add_repo | sudo bash -s - 

After that you can install i2pd as any other software package:

apt-get update apt-get install i2pd 

> > +
> > sudo systemctl enable i2pd
> > +
> > add to bitcoin.conf :
> > 
> > debug=i2p
> > #onlynet=i2p #optional
> > i2psam=127.0.0.1:7656
> > i2pacceptincoming=1

My name is @twofaktor 😆

VajraOfIndra, 2022-07-15:
> BTW: My name is @twofaktor laughing

Haha, ok, fixed! ;)

twofaktor, 2022-07-19:
> > BTW: My name is @twofaktor laughing
> 
> Haha, ok, fixed! ;)
> 

Error again, is @twofaktor 😁

VajraOfIndra, 2022-07-19:
> > > BTW: My name is @twofaktor laughing
> > 
> > 
> > Haha, ok, fixed! ;)
> 
> Error again, is @twofaktor 😁

Gee! Now it's fixed!... I think...! I'd better not write any PR rn.. it'd be full of typos! 😆

twofaktor, 2022-08-15:
New PR related! #1064

twofaktor, 2022-08-19:
PR #1064 merged! this issue can be closed!

closed 2022-08-19 by kristapsk