raspibolt/raspibolt #957: [FEATURE REQ] Incorrect screenshot and possibilities steps in the last version of Raspberry Pi Imager in "Operating system" section (Issue #954)

twofaktor, 2022-03-07:
### Describe
Create directly "admin" user instead "pi" user and set the password [A], include "Set locale settings" in “Advanced options” of Raspberry Pi Imager v1.7.1 and modify steps according to the correct instructions.

### Location in guide
https://raspibolt.org/guide/raspberry-pi/operating-system.html#configure-boot-options

Change the screenshot to this:
https://drive.google.com/file/d/1ZAwPeOmUaUnK3WtKEYxei34bMaPbfG4E/view?usp=sharing

### Additional context

- Modify: https://raspibolt.org/guide/raspberry-pi/remote-access.html#access-with-secure-shell

--> Change "pi" user to "admin" user login

- Modify https://raspibolt.org/guide/raspberry-pi/system-configuration.html#add-the-admin-user-and-log-in-with-it. 
--> Put first, steps:
- Create a new connection with the admin user
- Log in again using SSH (see [Access with Secure Shell](https://raspibolt.org/guide/raspberry-pi/remote-access.html#access-with-secure-shell) section), but now with the user “admin” and your password [A]
--> Modify title according to steps
--> Delete step: "Create a new user called “admin” with your password [A]"
--> Delete step: "Exit your current “pi” user session and exit SSH"

Stadicus, 2022-03-08:
Thanks, I think that would be a nice optimization of the core guide, simplifying the setup process. I'll reopen this issue to keep that in mind if someone wants to implement this.

visskiss, 2022-03-09:
Only issue with this is that for future upgrades or system restores, you will have different numbered users for all users, so that if you attempt to use rsync and retain owner and group, they will refer to the incorrect user. This is because users are numbers, not names. User admin will restore as bitcoin, user electrs will restore as bitcoin, etc...

VajraOfIndra, 2022-05-15:
This is still to be done and an important update to simplify the guide. Re-opening.

VajraOfIndra, 2022-07-15:
TBD

twofaktor, 2022-08-22:
Abandoned but there is an implementation thread on #1050. Proceed to close this problem

closed 2022-08-22 by twofaktor