raspibolt/raspibolt #958: Is “/mnt/static-channel-backup-external” listed?  nope 8-(

uskrusader, 2022-03-08:
Lightning/Channel Backup/Option 1: Local Backup/Set up a mounting point for the storage device

$ sudo mkdir /mnt/static-channel-backup-external
$ sudo chattr +i /mnt/static-channel-backup-external

done

[image]

[image]

I ran through this part of the guide twice, 
updated the /etc/fstab twice

Is “/mnt/static-channel-backup-external” listed?  
nope "...no such file or directory"
 
Any idea why it isnt listed?

in the meantime, I'll try using the remote github repositry method to backup.
cheers,

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4, 
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]

uskrusader, 2022-03-09:
[image]

VajraOfIndra, 2022-03-12:
> [image]

Ibelieve it's because the folder was created at: `/mnt/static-channel-backup-external`

You invoke `df` from the root folder of user "admin" (`/home/admin`) and you provide a relative path, therefore it's trying to find the folder at: `/home/admin/static-channel-backup-external` (which does not exist)

Instead, you need to either provide the absolute path: `df -h /mnt/static-channel-backup-external`
Or else, you need to move to the parent folder of the targeted folder first before using the relative path:
```
cd /mnt
df -h static-channel-backup-external
```

PR #983 (2022-04-15): LND Channel backup. Bug fixed. Motivated by Issue #958
#### What

Based on the implemented backup option with Dropbox, I wanted to switch the backup to the new variants. At the same time I also wanted to make a pull request to fix issue #958. I found a few more small bugs and fixed them.

Fixes #958 
Fixes #981

So I got the following error messages because the file channel.backup exists with a size of 0 bytes and root access rights:

admin@raspibolt:~ $ ls -al ~/.lnd/data/chain/bitcoin/mainnet/channel.backup
-rw-r--r-- 1 root root 0 Feb  1 09:58 /home/admin/.lnd/data/chain/bitcoin/mainnet/cha [... cut]
closed 2022-04-15 by VajraOfIndra