raspibolt/raspibolt #959: Can't scan QR code for Zeus with PuTTY

Stadicus, 2022-03-10:
### Describe the issue
Reported by @uskrusader

I'm stuck in the Zeus guide. The QR code is massive and I can't shrink it. Ctrl + - is not working in the putty termininal. It works for most of my other windows, but not putty where that ginormous QR code is. Cant fit it on screen 8-(

### Location in guide
https://raspibolt.org/guide/lightning/mobile-app.html#create-a-lndconnect-qr-code

Stadicus, 2022-03-10:
Yeah, that can be a bit of a pain. I have a curved screen and that also messes with the QR code display. Unfortunately, you can't copy/paste the text content of the code into Zeus, and you shouldn't need an online QR code generator as the code contains a secret.

I see two options: 

* generate a PNG with the QR code on the Raspberry Pi with the option `-o` and copy it to your local machine to scan:
  ```sh
  $ lndconnect --host=abcdefg..............xyz.onion --port=8080 -o
  > Wrote QR Code to file "lndconnect-qr.png"
  ```
* generate the underlying connection URL, copy/paste it to Windows, and use a local QR code generator to create a QR code
  ```sh
  $ lndconnect --host=abcdefg..............xyz.onion --port=8080 -o
  > lndconnect://abcdefg..............xyz.onion:8080?cert=MIICOD...FZsGuU
```

uskrusader, 2022-03-10:
> I see two options:
> 
> * generate a PNG with the QR code on the Raspberry Pi with the option `-o` and copy it to your local machine to scan:
>   ```shell
>   $ lndconnect --host=abcdefg..............xyz.onion --port=8080 -o
>   > Wrote QR Code to file "lndconnect-qr.png"
>   ```
> * generate the underlying connection URL, copy/paste it to Windows, and use a local QR code generator to create a QR code
>   ```shell
>   $ lndconnect --host=abcdefg..............xyz.onion --port=8080 -o
>   > lndconnect://abcdefg..............xyz.onion:8080?cert=MIICOD...FZsGuU
>   ```

ok, so I created the PNG on the raspberry, but I'm not sure how to get it from raspberry to my local machine to scan.

I plugged in an extra usb thumbdrive (called "SDC", I believe,  "SDA" is 1TB SSD and "SDB" is another usb thumbdrive for my lightning local SCB backup...[which is still not working, just using github repository instead for now])

I assume I use the cp or mv command to get the "lndconnect-qr.png" to the usb thumbdrive.  
I'm still pretty weak at this linux command-line stuff. 
I'm a recovering windows addict who just escaped the microsoft matrix. Thanks Morpheus! 

[image]

Stadicus, 2022-03-10:
To copy the file to the USB thumbdrive, you'd need to mount it first into the file system (here's [a tutorial](https://linuxconfig.org/howto-mount-usb-drive-in-linux))

Maybe it's easier to use WinSCP to copy the file over the network. Here's a very old part of the RaspiBolt guide which explains how to use it (back then, you had to copy the whole blockchain like that):
https://github.com/raspibolt/raspibolt/blob/raspibolt-v1-deprecated/raspibolt_50_mainnet.md#copy-using-winscp

uskrusader, 2022-03-10:
> To copy the file to the USB thumbdrive, you'd need to mount it first into the file system (here's [a tutorial](https://linuxconfig.org/howto-mount-usb-drive-in-linux))

THANKS!!  I used this mount guide and tinkered a bit.  I finally got the .png on the thumbdrive which made a beautiful QR code on my laptap.  Zeus is up and running!!  

I finished the Raspibolt guide!!  With zero prior linux/debian knowledge, it took 2 and a half months, very parttime.
Thank you much!

I still have a few small issues, but everything works over TOR.  Some of my local network connections and my local SCB backup need tinkering.  Learning...

Stadicus, 2022-03-11:
That's awesome to hear, I hope you enjoyed the ride! :tada:

closed 2022-03-11 by Stadicus

dawiepoolman, 2022-07-03:
hi guys, I followed exactly this process to get the QR code via a lndconnect-qr.png file but got stuck and raised this issue:
https://github.com/ZeusLN/zeus/issues/1014

did I miss something?