raspibolt/raspibolt #966: [BUG]  Incorrect command in mempool space install

visskiss, 2022-03-14:
### Describe the bug
Chown command is executed, but it should be chmod

sudo chown 600 /home/mempool/mempool/backend/mempool-config.json

should be 

sudo chmod 600 /home/mempool/mempool/backend/mempool-config.json

### Location in guide

[https://raspibolt.org/guide/bonus/bitcoin/mempool.html](https://raspibolt.org/guide/bonus/bitcoin/mempool.html)

VajraOfIndra, 2022-03-15:
Thanks @visskiss! Would you be interested to write a PR for the the issues you've idenitifed?

There are also issues #918 and #919 that concerns the Mempool guide. We could maybe do one PR that fixes the four issues at once maybe.. wdyt?

visskiss, 2022-03-15:
Yes, happy to assist. Do you want me to branch off master and then organize a pull request or do you have some other system (git flow?).

VajraOfIndra, 2022-03-16:
> Yes, happy to assist.

Thanks a lot! :)

> Do you want me to branch off master and then organize a pull request or do you have some other system (git flow?).

I don't think we'be been using git flow so far.. but I'll let @Stadicus answer you on this though as I'm very inexperienced with git.

PR #984 (2022-04-18): Several small bug fixes for Mempool bonus guide
#### What

Several bug fixes for the Mempool bonus guide:
- Fixes #918 : states the requirement of running Mempool with a minimum of 4 GB of RAM
- Fixes #965 : add a Tor proxy in Mempool config to fetch prices privately
- Fixes #966 : fixed command typo: chown -> chmod 
- a couple of minor wording changes for improved clarity

#### Scope

- [ ] significant change to core configuration
- [ ] independent bonus guide
- [x] simple bug fix
closed 2022-04-18 by VajraOfIndra