raspibolt/raspibolt #970: Latest release tag for Electrs doesn't match

xmready, 2022-03-16:
Following the guide to upgrade electrs, the output from `git describe --tags --abbrev=0` shows `v0.9.4` instead of the current version `v0.9.6`. The output of `git fetch` shows `v0.9.6` as a new tag. Am I in the wrong branch? Does `git checkout` still work if I use the new version number?

https://raspibolt.org/guide/bitcoin/electrum-server.html#for-the-future-electrs-upgrade

```
$ git fetch 
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 65 (delta 34), reused 45 (delta 25), pack-reused 0
Unpacking objects: 100% (65/65), 35.80 KiB | 191.00 KiB/s, done.
From https://github.com/romanz/electrs
 * [new branch]      dependabot/cargo/anyhow-1.0.56 -> origin/dependabot/cargo/anyhow-1.0.56
   5eb0701..6c68544  master                         -> origin/master
 * [new tag]         v0.9.6                         -> v0.9.6
 * [new tag]         v0.9.5                         -> v0.9.5
```

- Hardware platform: AMD Ryzen 3900x, 4GB RAM
- Operating system: Ubuntu Server 64bit LXC Container in Proxmox
- Version: Ubuntu 20.04.3 LTS

VajraOfIndra, 2022-04-12:
I believe this is an issue with the guide. The `git describe --tags --abbrev=0` does not output the version of the latest branch.

Instead of the present version:
```
$ git fetch
$ git describe --tags --abbrev=0
$ git checkout v9.9.9
```
I propose we use this:
```
$ git fetch
$ git reset --hard HEAD
$ git tag                                                 # this command will show all the tags, including the latest version to upgarde to
$ git checkout v9.9.9
```
@Stadicus @RooSoft Would that work well?

kristapsk, 2022-04-13:
I already noticed this month ago, see https://github.com/raspibolt/raspibolt/issues/951#issuecomment-1064384568.

VajraOfIndra, 2022-04-13:
> I already noticed this month ago, see [#951 (comment)](https://github.com/raspibolt/raspibolt/issues/951#issuecomment-1064384568).

This bug has been on my mind for a while. I believe we have the same issue with RTL or another guide like this. We really need to fix this asap.

Wdyt of the proposed commands just above? would that work well?

kristapsk, 2022-04-13:
This should work to get latest version tag:
```
admin ~/rust/electrs ₿ git tag | sort --version-sort | tail -n 1
v0.9.6
```

kristapsk, 2022-04-13:
I will go through Electrum server guide again, test this and open a PR if this works well for me.

PR #987 (2022-04-13): Bump Electrs version and update command to fetch latest release tag
#### What

Bump Electrs version to v0.9.6 and update command to fetch latest release tag for upgrade guide.

### Why

Resolves #951. Fixes #970.

#### Scope

- [ ] significant change to core configuration
- [ ] independent bonus guide
- [X] simple bug fix
closed 2022-04-13 by VajraOfIndra