raspibolt/raspibolt #973: New Laptop: How do I export SSH control from Windows putty to Linux SSD?

uskrusader, 2022-03-19:
My raspibolt is complete, but the windows computer I used to build it is being replaced.
I used windows and putty, but my new computer uses Ubuntu....and SSH 

New Laptop:
```
> 
> $ ssh pi@raspibolt.local
> The authenticity of host 'raspibolt.local (192.168.50.81)' can't be established.
> ECDSA key fingerprint is SHA256:1eGuC5xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxavE10.
> Are you sure you want to continue connecting (yes/no/[fingerprint])? y
> Please type 'yes', 'no' or the fingerprint: yes
> Warning: Permanently added 'raspibolt.local,192.168.50.81' (ECDSA) to the list of known hosts.
> pi@raspibolt.local: Permission denied (publickey).
```

I have three .txt files (keys?) from my old computer, but I don't know where or how to put them in the right place to make SSH work on my new linux laptop.

> authorized-Keys.txt,    pk,    and sshsk.ppk

zynos, 2022-03-20:
If you still have access to your old computer, you could temporarily allow login via password and then connect with your new ubuntu machine.

Otherwise this might help you:
https://askubuntu.com/questions/15378/how-do-i-install-an-ssh-private-key-generated-by-puttygen

uskrusader, 2022-03-20:
> If you still have access to your old computer, you could temporarily allow login via password and then connect with your new ubuntu machine.
> 

You mean, log in with old putty machine.....then remove the old keys?  ....then enable password?
Then login in with new computer/new password....then create new keys, ....then disable password?

this flow?

You don't think there is any way to import my old SSH keys into new Linux laptop?

zynos, 2022-03-20:
>Then login in with new computer/new password....then create new keys, ....then disable password?

 You dont need a new password for your admin account.
 But you need to enable the login via password to register your new ssh keys, so reverse this operation:
 https://raspibolt.org/guide/raspberry-pi/security.html#disable-password-login

 and after your new ssh keys work, disable password login again

uskrusader, 2022-03-20:
> You dont need a new password for your admin account. But you need to enable the login via password to register your new ssh keys, so reverse this operation: [...]

Thank you much!  I'm in on my new ubuntu laptop!
So I guess it is ok to have two sets of keys?  ...so both my new and old laptops can login.

that was much easier than i was expecting.  I didn't expect re-enabling & using the password would bypass the keys

closed 2022-03-20 by uskrusader