raspibolt/raspibolt #985: [BUG] LNBits install failure: pg_config executable not found

kristapsk, 2022-04-12:
### Describe the bug
When trying to [install LNBits](https://github.com/raspibolt/raspibolt/pull/977), at the `./venv/bin/pip install -r requirements.txt` step I get error:
```
...
Collecting psycopg2-binary==2.9.1
  Downloading psycopg2-binary-2.9.1.tar.gz (380 kB)
     |████████████████████████████████| 380 kB 10.6 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /home/lnbits/lnbits/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3q5hgzd5/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3q5hgzd5/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info
         cwd: /tmp/pip-install-3q5hgzd5/psycopg2-binary/
    Complete output (23 lines):
    running egg_info
    creating /tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info
    writing /tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txt
    writing manifest file '/tmp/pip-install-3q5hgzd5/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'
    
    Error: pg_config executable not found.
    
    pg_config is required to build psycopg2 from source.  Please add the directory
    containing pg_config to the $PATH or specify the full executable path with the
    option:
    
        python setup.py build_ext --pg-config /path/to/pg_config build ...
    
    or with the pg_config option in 'setup.cfg'.
    
    If you prefer to avoid building psycopg2 from source, please install the PyPI
    'psycopg2-binary' package instead.
    
    For further information please check the 'doc/src/install.rst' file (also at
    <https://www.psycopg.org/docs/install.html>).
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

`pg_config` is from PostgreSQL. Maybe some other bonus guide, which I haven't ran on my setup, introduces that as a dependency, or some other OS have it by default, and so other testers had it installed while it's missing on my machine?

### Location in guide
https://raspibolt.org/guide/bonus/lightning/lnbits.html

### Expected behavior
Install succeeds.

### Environment
 - Hardware platform: Odroid HC4
 - Operating system: Ubuntu Server 64bitt
 - Version: Ubuntu 20.04.4 LTS

VajraOfIndra, 2022-04-13:
> `pg_config` is from PostgreSQL. Maybe some other bonus guide, which I haven't ran on my setup, introduces that as a dependency, or some other OS have it by default, and so other testers had it installed while it's missing on my machine?

I tested it with the Raspberry Pi OS Lite (64-bit). RasPiOS Lite [does not contain this package[(https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/2022-01-28-raspios-bullseye-arm64-lite.info)..

I don't know how it might have been installed on my node..

kristapsk, 2022-04-13:
libffi and libpq are mentioned as dependencies in official [LNBits install guide](https://github.com/lnbits/lnbits-legend/blob/main/docs/devs/installation.md):
```
# install libffi/libpq in case "pipenv install" fails
# sudo apt-get install -y libffi-dev libpq-dev
```

kristapsk, 2022-04-13:
Ok, that one was caused by systemwide `setuptools` provided by Ubuntu being too old version. Solution is `./venv/bin/pip install setuptools --upgrade`, but then I got next error:
```
Collecting secp256k1==0.14.0
  Using cached secp256k1-0.14.0.tar.gz (2.4 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/lnbits/lnbits/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-lz0sop2n/secp256k1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-lz0sop2n/secp256k1/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-lz0sop2n/secp256k1/pip-egg-info
         cwd: /tmp/pip-install-lz0sop2n/secp256k1/
    Complete output (53 lines):
    0.29.1
    /home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
      warnings.warn(
    WARNING: The wheel package is not available.
      ERROR: Command errored out with exit status 1:
       command: /home/lnbits/lnbits/venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-p75uzxn0/pytest-runner/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-p75uzxn0/pytest-runner/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-w1951pdt
           cwd: /tmp/pip-wheel-p75uzxn0/pytest-runner/
      Complete output (12 lines):
      /home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
        warnings.warn(
      WARNING: The wheel package is not available.
      WARNING: The wheel package is not available.
      WARNING: The wheel package is not available.
      WARNING: The wheel package is not available.
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help
    
      error: invalid command 'bdist_wheel'
      ----------------------------------------
      ERROR: Failed building wheel for pytest-runner
    ERROR: Failed to build one or more wheels
    Traceback (most recent call last):
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/installer.py", line 82, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/home/lnbits/lnbits/venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpm2sn75ev', '--quiet', 'pytest-runner==2.6.2']' returned non-zero exit status 1.
    
    The above exception was the direct cause of the following exception:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-lz0sop2n/secp256k1/setup.py", line 265, in <module>
        setup(
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/__init__.py", line 86, in setup
        _install_setup_requires(attrs)
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/__init__.py", line 80, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/dist.py", line 861, in fetch_build_eggs
        resolved_dists = pkg_resources.working_set.resolve(
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/pkg_resources/__init__.py", line 789, in resolve
        dist = best[req.key] = env.best_match(
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1075, in best_match
        return self.obtain(req, installer)
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1087, in obtain
        return installer(requirement)
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/dist.py", line 941, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/home/lnbits/lnbits/venv/lib/python3.8/site-packages/setuptools/installer.py", line 84, in fetch_build_egg
        raise DistutilsError(str(e)) from e
    distutils.errors.DistutilsError: Command '['/home/lnbits/lnbits/venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpm2sn75ev', '--quiet', 'pytest-runner==2.6.2']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

kristapsk, 2022-04-13:
Ok, solution to that one is `./venv/bin/pip install wheel`.

PR #988 (2022-04-16): Fix missing / outdated deps for some OSes in LNBits guide
#### What

Fixes missing / outdated deps for some OSes (tested on Ubuntu Server 20.04.4 LTS) in LNBits guide.

#### Scope

- [ ] significant change to core configuration
- [X] independent bonus guide
- [ ] simple bug fix

Fixes #985.
closed 2022-04-16 by VajraOfIndra