raspibolt/raspibolt #1006: Question - Clearnet transfer to Tor

dawiepoolman, 2022-04-27:
Hi guys
I am in the process of upgrading an old setup of mine of a RaspiBolt 2 version from a raspberry pi 3B+ to a new Rpi4.  The old node had a static clearnet IP with open LND channels.  
Q: Can I move the channel backups to a new node when I configure Tor or will the channel backups not be compatible?

VajraOfIndra, 2022-04-29:
By channel backup, I assume you mean creating a new Tor node but re-using your existing LND wallet and channel DB. If so, I think (but it'd be good for someone else to confirm) that it's possible, as the clearnet vs Tor thing is at the networking level. 

However, you'll be losing the privacy benefits of using Tor since your public node will still be associated with that old clearnet IP address,..

The channel backup itself, the SCB, cannot be used to recreate a node. It is used to basically ask all your peers with whom you have a channel to force close them on their side for you to recover all your funds in the onchain LND wallets (asssuming all peers are online and do close their channel).

dawiepoolman, 2022-07-03:
thx guys, I managed to upgrade tor tor successfully with at least some of the channels still in place.
closing this issue.

closed 2022-07-03 by dawiepoolman