raspibolt/raspibolt #1016: PyBLOCK

SatoshiNakamotoBitcoin, 2022-05-03:
Open the terminal in Raspibolt.

$pip3 install pybitblock 
$pyblock

The First Menu appear
https://pbs.twimg.com/media/FQ3IoUWWYAA32CH.jpg

Select option A.

You are in.

Enjoy it.

About [PyBLOCK](https://github.com/curly60e/)

Stadicus, 2022-05-04:
Does PyBLOCK have any logs? 

I installed as mentioned above (although with "sudo") and went through the setup. All good.

Now, when I start `pyblock`, it just clears my screen:

https://user-images.githubusercontent.com/19550140/166642547-a2b69317-a2cf-45d2-b869-46ef91ae603e.mp4

SatoshiNakamotoBitcoin, 2022-05-04:
@curly60e 
Can you check this please ?

curly60e, 2022-05-04:
https://user-images.githubusercontent.com/55191248/166703347-0a1f6f03-b6aa-483d-b318-4b41c359efe1.mp4

SatoshiNakamotoBitcoin, 2022-05-04:
Thanks guys. 🙏

🧡

Stadicus, 2022-05-04:
Thanks for the quick reply. Now I'm glad I made the screencast, as it works fine. No changes needed... :)

Possible reason: Bitcoin Core was catching up on blocks (it was running on testnet for ~1 month, and I switched back shortly before installing PyBlock).

Stadicus, 2022-05-04:
> Also, if you execute pyblock in another folder it will generate new files, be aware of that.

Oh, that's very unexpected. Simply creating a `config` directory in the current location does not seem like a good practice. I never experienced this before. What's the reasoning behind that?

I think that needs at least a workaround in the form of a shell script that changes into a certain runtime directory, executes pyblock, and changes back to pwd. Otherwise, we'll just pollute the system with leftover config dirs.

curly60e, 2022-05-04:
Yeah, I didn't find out how to setup a base folder that pyblock could reach there for config files. If you have some tips on how to do it I will check that out and improve it.

Glad you could make it work.

closed 2022-07-10 by github-actions[bot]